MiczFlor/RPi-Jukebox-RFID #1554: No webapp on future3

Groovylein, 2021-10-09:
### Describe your problem 

Webapp does not appear.

#### What's your hardware set up? 

RPi 3 B+

#### CMD output:
```
pi@MusicPi:~/RPi-Jukebox-RFID/src/webapp $ npm run build

> webapp@0.1.0 build
> react-scripts build

Creating an optimized production build...

<--- Last few GCs --->

[721:0x5943638]   373029 ms: Mark-sweep (reduce) 226.1 (236.8) -> 225.4 (236.8) MB, 4835.0 / 0.3 ms  (average mu = 0.171, current mu = 0.040) allocation failure scavenge might not succeed
[721:0x5943638]   377912 ms: Mark-sweep (reduce) 226.3 (237.0) -> 225.6 (237.3) MB, 4843.4 / 0.3 ms  (average mu = 0.094, current mu = 0.008) allocation failure scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory
```

Groovylein, 2021-10-09:
I've run the commands manually from the install script:
```
cd ${INSTALLATION_PATH}/src/webapp
npm ci --prefer-offline --no-audit --production
rm -rf build
npm run build
```

Groovylein, 2021-10-09:
The install script did fail as well. 

I will set this option and try again 😊

pabera, 2021-10-09:
I updated the UI lib to the most recent version just before release and boom: that issue started to occur on my RPi 3+ as well.

That's why I added the `NODE_OPTIONS` within the install script. Maybe it does not set it right. I  briefly added a log output.

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/02d5fe89f880e607b2e7464dd9b3fe4846e860a6/installation/install-jukebox.sh#L296-L306

If you don't mind, you could retry the procedure. If it fails again, you could check the `INSTALL_XXX.log` file and report which options was set during the installation. Chose the `development branch` options during install. The commit has not been merged to the release branch yet.

Could you be a bit more specific about your SD card

Groovylein, 2021-10-10:
With the ENV variable building the Webapps was successful.

I will try a new installation from a fresh Raspbian. Let's see if this works