MiczFlor/RPi-Jukebox-RFID #1613: playermd  **prev** and **next** methods are causing an exception

arne123, 2021-11-02:
### Describe your problem 

playermd  **prev** and **next** methods are causing an exception (via mpd), when no song is played.
This is very likly to happen since this functions are often used with buttons.

#### If possible, try to attach logs from ... (paths from RPi)
02.11.2021 21:46:07 -  766:plugs.py           - jb.plugin            - Dummy-3         - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 758, in call_ignore_errors
    result = _call(package, plugin, method, args=args, kwargs=kwargs, as_thread=as_thread, thread_name=thread_name)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 706, in _call
    return func(*args, **kwargs)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/playermpd/__init__.py", line 316, in prev
    self.mpd_client.previous()
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 435, in mpd_command
    return wrapper(self, name, args, callback)
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 498, in _execute
    return retval()
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 420, in command_callback
    res = function(self, self._read_lines())
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 353, in _parse_nothing
    for line in lines:
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 552, in _read_lines
    line = self._read_line()
  File "/usr/local/lib/python3.7/dist-packages/mpd/base.py", line 541, in _read_line
    raise CommandError(error)
mpd.base.CommandError: [55@0] {} Not playing

varac, 2024-04-06:
Can this get closed as duplicate of #2294 ?

s-martin, 2024-04-07:
Closing. Issue is tracked thoroughly in #2294