MiczFlor/RPi-Jukebox-RFID #1618: RFID doens´t work anymore, even with fresh installation

timoweber1985, 2021-11-06:
## Bug

### What I did

I started the PI and moved the RFID Card over the Reader and nothing happened. Days before it worked without any issue (no change). I decided to reinstall the PI (OS and JukeBox Software completely), but during the installation it throughs error where I selected the card reader. I tried it again and again with RegisterDevice.py (and all other files that are available). I read all issues in this forum and tried a lot, but at the end, the device cannot be registered without this error and the RFID Number will be not display and used in the Jukebox. 

`pi@raspberrypi:~ $ /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py 
Choose the reader from list
0 Logitech M705
1 HID 413d:2107
2 MFRC522
3 RDM6300
4 PN532
Device Number: 1
Exception ignored in: <function InputDevice.__del__ at 0xb60df3d8>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 160, in __del__
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 305, in close
  File "/usr/local/lib/python3.7/dist-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.7/asyncio/events.py", line 640, in get_event_loop
  File "/usr/lib/python3.7/asyncio/events.py", line 660, in new_event_loop
  File "/usr/lib/python3.7/asyncio/unix_events.py", line 51, in __init__
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 49, in __init__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 373, in __init__
  File "/usr/lib/python3.7/asyncio/coroutines.py", line 28, in _is_debug_mode
AttributeError: 'NoneType' object has no attribute 'dev_mode'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb613da08>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function InputDevice.__del__ at 0xb60df3d8>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 160, in __del__
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 305, in close
  File "/usr/local/lib/python3.7/dist-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.7/asyncio/events.py", line 640, in get_event_loop
  File "/usr/lib/python3.7/asyncio/events.py", line 660, in new_event_loop
  File "/usr/lib/python3.7/asyncio/unix_events.py", line 51, in __init__
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 49, in __init__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 373, in __init__
  File "/usr/lib/python3.7/asyncio/coroutines.py", line 28, in _is_debug_mode
AttributeError: 'NoneType' object has no attribute 'dev_mode'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb613da08>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
`
It is somehow the same like: 🐛 | Installation fails on RegisterDevice.py #1501 

### What happened

Installation not 100% succeeded and Jukebox with RFID Reader is not working. 

### I expected this to happen

That it starts playing music

### Further information that might help

Analytics:
`************************************
*** PHONIEBOX INFO
*** version: 2.3 - 2425890 - master
*** edition: plusSpotify
*** Audio_iFace_Name: Master
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
[... 622 lines cut ...]
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

`master`

### Installscript

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
`

## Hardware

### RaspberryPi version

Raspberry Pi 4 Model B Rev 1.1

### RFID Reader

HID 413d:2107

timoweber1985, 2021-11-06:
Short Update: I already tried another RFID Reader and same issue. It looks not like an RFID Reader Issue, more like a OS or Python issue.

danielperna84, 2021-11-27:
Not sure if this is actually all that is needed, but closing the devices after the name has been written to `deviceName.txt` gets rid of the exception. To do this, append the following to the `RegisterDevice.py` script:

```python3
for dev in devices:
    dev.close()
```

This alone does not make my RFDI-reader work again (after upgrading the OS and from 2.2 to 2.3) for me though.

johkoenig, 2021-12-22:
Hello everybody,
I am also facing some issues. Today, I've set up a new phoniebox.
For me, the following solution worked:

1. Install a missing python package: `sudo apt-get install python3-evdev -y`
2. Register the device: python3 `/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py`
3. Kill the script with `Ctrl + C` after a while
4. Activate RFID on the web interface

I assume that step 1 is a error in the install script while step 3 is a error in the RegisterDevice script. 

Best regards,
jk

pabera, 2021-12-23:
Are you using Bullseye (new Debian version) or Buster? Version 2.3 is currently only compatible with Buster.

johkoenig, 2021-12-23:
Yes, the new installation was using bullseye. I didn't know it wasn't supported. Apart from the workaround given above, everything works fine, though.

closed 2022-03-24 by s-martin

EarMaster, 2023-03-11:
> Not sure if this is actually all that is needed, but closing the devices after the name has been written to `deviceName.txt` gets rid of the exception. To do this, append the following to the `RegisterDevice.py` script:
> 
> ```python
> for dev in devices:
>     dev.close()
> ```

This solution worked for me too.
I was upgrading from 2.3.0 on a Rasberry Pi 4 running buster and after the upgrade the RFID reader wasn't working any more.

mhawkshaw, 2023-04-02:
> Hello everybody, I am also facing some issues. Today, I've set up a new phoniebox. For me, the following solution worked:
>
> 1. Install a missing python package: `sudo apt-get install python3-evdev [...]

I had the same problem and installing python3-evdev fixed it for me. I'm running Buster.