MiczFlor/RPi-Jukebox-RFID #1619: Getting RC522 reader to work

alexanderkoller, 2021-11-07:
Hi all,

I'm trying to use RPi-Jukebox-RFID with an AZ-Delivery RC522 reader (have tried both Raspberry Pi 3A+ and Zero 2 W), but the reader just won't read the RFID chip and card that came in the package.

This is both from within RPi-Jukebox-RFID (installed with the install script) and with the test script in ondryaso/pi-rc522. The RPi-Jukebox web interface shows the RFID reader as available. But in either case, when I hold one of the chips or cards that came with the reader near it, nothing happens.

I am using the [wiring with IRQ to GPIO24](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#hardware-with-fixed-pin-assignment), and have double-checked that all wires are connected correctly.

I am using the current version of Raspberry Pi OS, and have enabled SPI in raspi-config. The kernel modules are loading:
```
$ uname -a
Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

$ lsmod|grep spi
spidev                 20480  0
spi_bcm2835            20480  0

$ ls -la /dev/spi*
crw-rw---- 1 root spi 153, 0 Nov  7 09:42 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Nov  7 09:42 /dev/spidev0.1
```

Any idea what I could try to make this work? Or how I would even debug this?

Alternatively, where can I get an RFID reader that works reliably with the RPi Jukebox? The Neuftech reader seems to be sold out on amazon.de.

lenlennart, 2021-11-08:
The RC522 reader is working very reliably for me.
Please ensure you follow the wiring described here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader
As it seems you already tried a different pi and even re-installing (?) - i suspect either your soldering is faulty or your reader is broken...

alexanderkoller, 2021-11-09:
Huh, turns out that that the problem was caused by my GPIO extension board. This despite the fact that I measured all the connections in it, and they are all fine.

At any rate, I can confirm that the RC522 works fine. Closing the issue.

closed 2021-11-09 by alexanderkoller