MiczFlor/RPi-Jukebox-RFID #1623: Installation Errors on RPI Zero2

zimbo86, 2021-11-11:
## Bug

### What I did

Trying to install RPi-Jukebox-RFID on Raspberry PI Zero2

### What happened

Installation throws errors running 
`sudo python3 -m pip install --upgrade --force-reinstall -q -r  RPi-Jukebox-RFID/requirements.txt`

```

    ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-gscmdvmf
       cwd: /tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/
  Complete output (87 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.9
  creating build/lib.linux-armv7l-3.9/RPi
  copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
  creating build/lib.linux-armv7l-3.9/RPi/GPIO
  copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
  running build_ext
  building 'RPi._GPIO' extension
  creating build/temp.linux-armv7l-3.9
  creating build/temp.linux-armv7l-3.9/source
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-armv7l-3.9/source/event_gpio.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-armv7l-3.9/source/py_gpio.o
  source/py_gpio.c: In function ‘PyInit__GPIO’:
  source/py_gpio.c:1046:4: warning: ‘PyEval_ThreadsInitialized’ is deprecated [-Wdeprecated-declarations]
   1046 |    if (!PyEval_ThreadsInitialized())
        |    ^~
  In file included from /usr/include/python3.9/Python.h:145,
                   from source/py_gpio.c:23:
  /usr/include/python3.9/ceval.h:129:36: note: declared here
    129 | Py_DEPRECATED(3.9) PyAPI_FUNC(int) PyEval_ThreadsInitialized(void);
        |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
  source/py_gpio.c:1047:7: warning: ‘PyEval_InitThreads’ is deprecated [-Wdeprecated-declarations]
   1047 |       PyEval_InitThreads();
        |       ^~~~~~~~~~~~~~~~~~
  In file included from /usr/include/python3.9/Python.h:145,
                   from source/py_gpio.c:23:
  /usr/include/python3.9/ceval.h:130:37: note: declared here
    130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
        |                                     ^~~~~~~~~~~~~~~~~~
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/py_pwm.c -o build/temp.linux-armv7l-3.9/source/py_pwm.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/soft_pwm.c -o build/temp.linux-armv7l-3.9/source/soft_pwm.o
  arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-z,relro -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.9/source/c_gpio.o build/temp.linux-armv7l-3.9/source/common.o build/temp.linux-armv7l-3.9/source/constants.o build/temp.linux-armv7l-3.9/source/cpuinfo.o build/temp.linux-armv7l-3.9/source/event_gpio.o build/temp.linux-armv7l-3.9/source/py_gpio.o build/temp.linux-armv7l-3.9/source/py_pwm.o build/temp.linux-armv7l-3.9/source/soft_pwm.o -o build/lib.linux-armv7l-3.9/RPi/_GPIO.cpython-39-arm-linux-gnueabihf.so
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:41: multiple definition of `module_setup'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:41: first defined here
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:40: multiple definition of `setup_error'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:40: first defined here
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:39: multiple definition of `rpiinfo'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:39: first defined here
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:38: multiple definition of `gpio_direction'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:38: first defined here
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:37: multiple definition of `pin_to_gpio'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:37: first defined here
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:36: multiple definition of `pin_to_gpio_rev3'; build/temp.linux-armv7l-3.9/s [... cut ...]
  /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:41: multiple definition of `module_setup'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-ryf7keem/rpi-gpio_510ada7fb31841b08079ad0a8ed04d0a/source/common.h:41: first defined here
[... 35 lines cut ...]
  ERROR: Failed building wheel for RPi.GPIO

``

```

## Software

### Base image and version

`Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`

### Branch / Release

master

### Installscript

buster-install-default.sh

## Hardware

### RaspberryPi version

No peripheral devices connected to RPI.

zimbo86, 2021-11-11:
oooookay -I think i solved my issue!

I´m not that deep into Linux and RPI, however I noticed additionally to the posted error samba configuration was not succesfull.
This happens because php 7.3 is hardcoded in buster-install-default.sh.
I replaced "7.3" by "7.4" (which is the only php version present on current RaspiOS) so the script runs till the end and the web server is available. The outlined error still occurs, however all services (Reader, Buttons, etc.) seem to work fine.
Also I had to install 
sudo apt install python3-evdev
to get my RC522 to work.

XTNs6QTTm5hf, 2021-11-13:
Hi,

thanks for the hint about python3-evdev.
But are you sure you installed only python3-evdev to get the RC522 reader working on a fresh raspbian bullseye installation?

I just tried to install phoniebox on a new pi zero 2 running bullseye. I have replaced the two 7.3 occurences in install script with 7.4. I was running into the same error messages your where printing in your initial comment. Phoniebox itself starts up without issues. I have the webinterface and sound up and running. Also spotify seems to be working, but I did not manage to get the RFID reader RC522 to work. 

RC522 was running fine on an old Pi zero 1, so wiring should still be fine.

zimbo86, 2021-11-14:
@XTNs6QTTm5hf i did the following steps:

`sudo python3 -m pip install --upgrade --force-reinstall py532lib
sudo python3 -m pip install -q -r RPi-Jukebox-RFID/components/rfid-reader/RC522/requirements.txt
sudo apt install python3-evdev
cd ~/RPi-Jukebox-RFID/scripts
cp Reader.py.experimental Reader.py
python3 RegisterDevice.py
==> Select RC522 (in my case number 1)
sudo systemctl restart phoniebox-rfid-reader.service
sudo raspi-config
==> Interfacing Options, Enable SPI
sudo reboot
`

after that I had to enable RFID in webinterface once.
Hope that helps.

XTNs6QTTm5hf, 2021-11-14:
Hi,

thanks a lot. I forgot to enable the SPI interface on fresh installed bullseye using raspi-config. Now it is working fine.

I also got rid of the initial error messages you where also encountering. Seems to be something related to new gcc version used by bullseye. 

You have to run the `python3 -m pip install` using CFLAGS="-fcommon" and all the error messages are not showing  up any more. So e.g.
`sudo CFLAGS="-fcommon" python3 -m pip install --upgrade --force-reinstall -q -r RPi-Jukebox-RFID/requirements.txt`

Will try to create an updated install script for bullseye later on containing all fixes mentioned in this thread:

1) Replace hard coded php version string "7.3" with "7.4"
2) Install python3-evdev
3) run "pip install" using CFLAGS mentioned above

Thanks again

XTNs6QTTm5hf, 2021-11-14:
Hi again,

attached a first try of a bullseye install script. Unfortunately my second SD card lying around seems to be broken, so it is still untested, but maybe a start for anyone else. Will try again during next days to verify.

[bullseye-install-default.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/7534689/bullseye-install-default.zip)

pabera, 2021-11-14:
If you are brave enough, you could try [V3](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-) 😇

XTNs6QTTm5hf, 2021-11-15:
> If you are brave enough, you could try [V3](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-) innocent

Good hint, but first need to get V2.X with spotify and GPIO buttons running on bullseye pi zero2 for my kids. Afterwards I am happy to try out V3 on my spare pi zero 1 :-)

XTNs6QTTm5hf, 2021-11-15:
> Hi again,
>
> attached a first try of a bullseye install script. Unfortunately my second SD card lying around seems to be broken, so it is still untested, but maybe a start for anyone else. Will try [...]

I just tested the bullseye installation script on a fresh bullseye SD Card and it is working fine. Together with the fix mentioned in [issue 1309](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1309#issuecomment-948510269) everything seems to work fine now.

And compared to a pi zero 1 the new pi zero 2 is incredible. On old setup it took quite some time between pressing a GPIO button and the response. Now on pi zero 2 GPIO button actions are immediately executed. At least I guess improvements are mainly coming from quad core CPU and not from bullseye upgrade

zimbo86, 2021-11-15:
> I just tested the bullseye installation script on a fresh bullseye SD Card and it is working fine. Together with the fix mentioned in [issue [...]

Can confirm that the performance of Zero2 is pretty good. Way better than my test Pi 2B and 1/4 the price of an RPI4. As its running headless for me its the best option atm!

closed 2021-12-02 by pabera