MiczFlor/RPi-Jukebox-RFID #1630: 🐛 | BUG SUMMARY

Thilo-90, 2021-11-17:
## Bug

### What I did
I tried to setup Phoniebox with an RC522 reader and could not get it to work.
Wired up the RC522 (it is working, tested with [PyMyLifeUp](https://pimylifeup.com/raspberry-pi-rfid-rc522/)) and also double checked the GPIO and Pins on the Reader.

So to provide clear steps i formatted everything and started again.

Installed latest Raspberry Pi OS using Raspberry Pi Imager V1.6.2

I enabled SSH and ran the one-line install command which guided me through the installation process

### What happened

I already got error messages during the installation but did not pay too much attention the first time

```#####################################################
#
# START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes
# between 15 minutes to half an hour, depending on
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] y
################################################
WIFIconfig="YES"
WIFIcountryCode="DE"
WIFIssid="WLAN-VSBG5D"
WIFIip="192.168.2.108"
WIFIipRouter="192.168.2.1"
EXISTINGuse=NO
AUDIOiFace="Headphone"
SPOTinstall="NO"
MPDconfig="YES"
DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
GPIOconfig="YES"
################################################
Generating locales (this might take a while)...
  en_GB.UTF-8... done
Generation complete.
Selecting previously unselected package apt-transport-https.
(Reading database ... 99541 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_2.2.4_all.deb ...
Unpacking apt-transport-https (2.2.4) ...
Setting up apt-transport-https (2.2.4) ...
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
Selecting previously unselected package libspotify12.
(Reading database ... 99545 files and directories currently installed.)
Preparing to unpack .../libspotify12_12.1.103-0mopidy1_armhf.deb ...
Unpacking libspotify12 (12.1.103-0mopidy1) ...
Selecting previously unselected package libspotify-dev.
Preparing to unpack .../libspotify-dev_12.1.103-0mopidy1_armhf.deb ...
Unpacking libspotify-dev (12.1.103-0mopidy1) ...
Setting up libspotify12 (12.1.103-0mopidy1) ...
Setting up libspotify-dev (12.1.103-0mopidy1) ...
Processing triggers for man-db (2.9.4-2) ...
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 99712 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1%3a1.20211029-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1:1.20211029-1) ...
Setting up raspberrypi-kernel-headers (1:1.20211029-1) ...
Preconfiguring packages ...
(Reading database ... 154403 files and directories currently installed.)
[... 1377 lines cut ...]
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

`master`

### Installscript

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

## Hardware

### RaspberryPi version

`Raspberry Pi 4 Model B Rev 1.4`

### RFID Reader

i.e. `MAKEVMA405`
[manual ](https://asset.conrad.com/media10/add/160267/c1/-/en/001612802ML01/gebruiksaanwijzing-1612802-maker-factory-rfid-schrijf-en-leesmodus-module-compatibel-met-arduino.pdf)

Lippsson, 2021-11-17:
I have the same issue like Thilo-90. I tried on a raspbery zero and raspberry 4. Same thing here. I had a running Phoniebox system on a rasperry 3. When i put in that SD card to the zero or rp4 it works well. But i still cant install this script on a fresh uptodate system

Thilo-90, 2021-11-17:
Thanks for the update Lippsson, though it won't work as intended for me.
Please see below my input / output

```
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py
Choose the reader from list
0 vc4
1 vc4
2 MFRC522
3 RDM6300
4 PN532
Device Number: 2

```

If i hit Enter after typing the number, nothing happens. The script does not proceed for me. I have to close the console and start a new ssh connection, though the same thing happens.

I also tried a reboot to see if it is working, but it is not. Neither the script nor the reader (miracles happen? sometimes?)

Lippsson, 2021-11-17:
Hi Thilo!
Yes the same happend on my side. I pressed the number and nothing. Then i closed with Ctrl+C and get to the next line 
$ sudo systemctl restart phoniebox-rfid-reader.service

Thilo-90, 2021-11-18:
Hi Lippsson, thank you so much. Just tested it and it finally works. :)

closed 2021-11-18 by Thilo-90

zimbo86, 2021-11-21:
hmm I also struggling with my RC522 after a new installation.
All seems running OK but the reader does not read.
Can you confirm that there is a 
`RuntimeWarning: This channel is already in use, continuing anyway.`
in 
` journalctl -f`
after you run 
`sudo systemctl restart phoniebox-rfid-reader.service`
?

Is there another way to test RC522 (the Read.py Write.py does not work anymore since it is Pythons2)

Thilo-90, 2021-11-21:
Hi @zimbo86, 

i just ran the commands you mentioned. Here is my log

```
pi@raspberrypi:~ $ sudo systemctl restart phoniebox-rfid-reader.service
pi@raspberrypi:~ $ journalctl -f
-- Journal begins at Sat 2021-10-30 12:36:50 BST. --
Nov 21 16:02:01 raspberrypi sudo[5612]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Nov 21 16:02:01 raspberrypi systemd[1]: Stopping Phoniebox RFID-Reader Service...
Nov 21 16:02:01 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Succeeded.
Nov 21 16:02:01 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Nov 21 16:02:01 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Consumed 9min 6.021s CPU time.
Nov 21 16:02:01 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Nov 21 16:02:01 raspberrypi sudo[5612]: pam_unix(sudo:session): session closed for user root
Nov 21 16:02:02 raspberrypi daemon_rfid_reader.py[5615]: /usr/local/lib/python3.9/dist-packages/pirc522/rfid.py:90: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
Nov 21 16:02:02 raspberrypi daemon_rfid_reader.py[5615]:   GPIO.setup(pin_rst, GPIO.OUT)
Nov 21 16:02:02 raspberrypi daemon_rfid_reader.py[5615]: 2021-11-21 16:02:02,053 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
```

My Tags are read without any problem though by now. Also after restarting the service.

zimbo86, 2021-11-22:
> Hi @zimbo86,
>
> i just ran the commands you mentioned. Here is my log
>
> ```
> pi@raspberrypi:~ $ sudo systemctl restart phoniebox-rfid-reader.service
> pi@raspberrypi:~ $ journalctl -f
> -- Journal begins at [...]

Got a soldering mistake - reader is detected now ;)