MiczFlor/RPi-Jukebox-RFID #1631: 🐛 | NAS integration - Web Application very long loading times

Thilo-90, 2021-11-19:
## Bug
I mounted my "Multimedia Folder" from my Synology NAS to the shared/audiofolders.
Depending on the folder i mount (see below section _naming and editing mount file_ `WHAT=path`) the loading times of the Webserver Application increase drastically.

Eg. if i mount `folder a` it is not noticable.
If i mount one `folder 1` it is noticable.
If i mount the whole `multimedia` folder the Web App is loading "forever".

_simplified_
```
multimedia
---folder 1
  |---folder a
      |---audio1.mp3
      |---audio2.mp3
  |---folder b
  |---folder c
---folder 2
---folder 3
```

Since i am still setting everything up, there is only one Webradio Stream currently linked to an RFID Card, and that can be played / stopped during the loading time. It is just the webserver which is not showing, which makes Registering Cards to the NAS not very practical.

### What I did
**Mounting process of my NAS**
```
sudo nano /etc/smbcredentials
################################################
username=xxx
password=yyy
################################################
```

**changing access rights** 
```
sudo chown root:root /etc/smbcredentials
sudo chmod 600 /etc/smbcredentials
```

**naming and editing mount file** (had to use double escape `\\` for the standard audiofolders location)
```
sudo nano /etc/systemd/system/home-pi-RPi\\x2dJukebox\\x2dRFID-shared-audiofolders-NAS.mount
################################################
[Unit]
Description=Mount_der_Synology
Requires=serverctl.service
After=serverctl.service
Conflicts=shutdown.target
ConditionPathExists=/home/pi/RPi-Jukebox-RFID/shared/audiofolders/NAS

[Mount]
What=//192.168.2.123/multimedia/
Where=/home/pi/RPi-Jukebox-RFID/shared/audiofolders/NAS
Options=credentials=/etc/smbcredentials,uid=1000,gid=1000
Type=cifs

[Install]
WantedBy=multi-user.target
################################################
```

**setting serverctl.service**
```
sudo nano /etc/systemd/system/serverctl.service
################################################
[Unit]
Description=serverctl.service:   Waiting for Network or Server to be up
After=network.target

[Service]
Type=oneshot
TimeoutStartSec=95
ExecStart=/usr/local/sbin/serverctl 192.168.2.123

[Install]
WantedBy=multi-user.target
################################################
```

**setting bash script**
```
sudo nano /usr/local/sbin/serverctl
################################################
#!/bin/bash
#==============================================================================================
# serverctl by TomL*thlu.de
#
# Script-Name :  serverctl
# Version     :  1.0
# Date        :  31.07.2017
# Lizenz      :  GNU GPL3
# Description :  Check if given server is reachable
#==============================================================================================

[ -z "$1" ] && Server="8.8.8.8" || Server=$1
echo "active/running   Server=$Server" | systemd-cat -t "thlu:`basename $0`" -p "info"

timeout=85
Diff=0
HomeNetIsConnect=-1

Start=$(date +%s);

while [ true ]; do
    /bin/ping -c1 -W1 -q $Server &>/dev/null
    HomeNetIsConnect=$?

    [ $HomeNetIsConnect -eq 0 ] && break
    /bin/sleep 0.5

    End=$(date +%s);
    Diff=$((End-Start))
    [[ Diff -gt timeout ]] && break
done
rc=0
if [[ $HomeNetIsConnect -eq 0 ]]; then
    echo "Host $Server is reachable! (RC:$HomeNetIsConnect, after $Diff Seconds wait)" | systemd-cat -t "thlu:`basename $0`" -p "info"
else
    echo "Host $Server is not reachable! (RC:$HomeNetIsConnect, after $Diff Seconds wait)" | systemd-cat -t "thlu:`basename $0`" -p "err"
    rc=1
fi

echo "Successful terminated with exitcode=$rc" | systemd-cat -t "thlu:`basename $0`" -p "info"
exit $rc
################################################
```

**setting access rights**
```
sudo chown root:root /usr/local/sbin/serverctl /etc/systemd/system/serverctl.service /etc/systemd/system/home-pi-RPi\\x2dJukebox\\x2dRFID-shared-audiofolders-NAS.mount
sudo chmod 644 /etc/systemd/system/serverctl.service /etc/systemd/system/home-pi-RPi\\x2dJukebox\\x2dRFID-shared-audiofolders-NAS.mount
sudo chmod 755 /usr/local/sbin/serverctl
```

**mounting the NAS**
`sudo systemctl start home-pi-RPi\\x2dJukebox\\x2dRFID-shared-audiofolders-NAS.mount`

### What happened

see Bug description

### I expected this to happen
Smooth NAS integration and Webserver without long loading times so i can easily link Folders to RFID cards

### Further information that might help
On additional notice, i usually cannot umount and have to reboot, thats why i currently only mount manually and not automatically on startup
```
sudo umount /home/pi/RPi-Jukebox-RFID/shared/audiofolders/NAS
umount: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/NAS: target is busy.
```

## Software

### Base image and version
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
`master`

### Installscript
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

## Hardware

### RaspberryPi version

`Raspberry Pi 4 Model B Rev 1.4`

pabera, 2021-11-19:
I know @Luegengladiator has done something in this direction. Maybe he can help.

Luegengladiator, 2021-11-19:
@pabera yes you are right. 
Let's see what I can do on that.

I tried something similar but struggled with the CIFS mount as well. 
So I switched to NFS. Here are the corresponding entries of my /etc/fstab:

<SERVERIP>:/PATH/to/RFIDs /home/pi/RPi-Jukebox-RFID/shared/shortcuts nfs defaults 0 0
<SERVERIP>:/PATH/to/Music /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ nfs defaults  0  0
<SERVERIP>:/PATH/to/Audiobooks /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Audiobooks nfs defaults 0 0

This one is mounted during boot due to the fstab entry before starting other services like Webserver.
My structure is way more complex than yours but I didn't find any performance problems. 
At the moment 3 Pi's using the same structure so I have to place the files only once. (shareable RFID-cards)

When using NFS v4, the integration is a little different.
NFSv4 introduces the concept of pseudo file system, which provides a single point of entry “/” (root) to 
access all the file systems below that entry point. This is similar to but not the same as the procfs found in 
other Linux implementations of pseudo file system. 
In [RFC 3530](https://www.ietf.org/rfc/rfc3530.txt), the following is mentioned regarding the NFSv4 server implantation of pseudo file system: 
```
NFS version 4 servers avoid this name space inconsistency by presenting all the exports within 
the framework of a single server name space. An NFS version 4 client uses LOOKUP and READDIR 
operations to browse seamlessly from one export to another. Portions of the server name space 
that are not exported are bridged using a "pseudo filesystem" that provides a view of exported 
directories only.  A pseudo filesystem has a unique fsid and behaves like a normal, read only 
filesystem
```

If possible, please try integrating the desired folder using NFS v3 (v4 taking into account the comments). If there are reasons for using CIFS, you have to look more closely at the configuration of the servers (authorizations, USER mapping, etc.)

Thilo-90, 2021-11-20:
Hi, there is actually no given reason for me to use CIFS.

It is just, that i am a total noob when it comes to setting up such stuff. I am a dumb Windows user and almost never work with linux console commands. Therefor, CIFS was just the first tutorial i found :)

Ok, i can somewhat follow what you proposed.
I am trying to set things up accordingly and read a bit about using nsf for mounting.

I enabled nsf on my Synology for the `Multimedia` folder and added the IP of my Phoniebox.

I tried in fstab but could not get it to work immediatly so i wanted to try mounting manually to check the problems.

```
showmount -e 192.168.2.123
Export list for 192.168.2.123:
/volume1/Multimedia 192.168.2.108

sudo mount -t nfs -vvvv -o nfsvers=3 192.168.2.123:volume1/Multimedia/Hörspiele/ /home/pi/RPi-Jukebox-RFID/shared/audiofolders/NAS/
mount.nfs: timeout set for Sat Nov 20 10:10:07 2021
mount.nfs: trying text-based options 'nfsvers=3,addr=192.168.2.123'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.2.123 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.2.123 prog 100005 vers 3 prot UDP port 892
mount.nfs: mount(2): Permission denied
mount.nfs: access denied by server while mounting 192.168.2.123:volume1/Multimedia/Hörspiele/

```
For handling this error i found [this tutorial](https://www.thegeekdiary.com/mount-nfs-access-denied-by-server-while-mounting-how-to-resolve/).

But now i am kind of lost. I read about the `/etc/exports` file but i do not really understand how to set that up.
Would be really glad if i could receive some help with setting that one up (if thats the missing step)

Luegengladiator, 2021-11-20:
Hi @Thilo-90 , 
no problem. We all started once. So applause for you to get this working that fast.
About the fstab and the Webserver I have to have a deeper look on the actual Version. 
Maybe there are some "hidden" obstracles which I'm not aware of. 
I'm asuming you are using the default setup of phonie? No special Webserver nor some other "special" setup which we should be aware of?
Kind regards
Luegengladiator

Thilo-90, 2021-11-20:
regarding fstab and getting it to mount during boot, i guess that would be possible if i can delay the mounting. I somewhere read about that part.
Though the long loading times are still a problem. 
we are talking about around 200gb of audiobooks. i guess Phoniebox wants to index all folders for displaying them in the webapp which might be too much.

On another note, i thought about avoiding to mount the NAS itself but instead making use of the UPNP/DLNA feature of shared folders. 
I uploaded one test file in the shared music folder and opened it in windows with my browser without any problems.
Though on my smarphone and on the phoniebox it is not working.
`file://192.168.2.123/music/test/test.mp3`
I am not sure if this is a permissions problem with my Synology preventing access.

If i use a dedicated mediaplayer for UPNP/DLNA i can see the folder and file and play it without any access problems.

Though even if i get this method working, that would mean a lot of streaming folders in the `shared/audiofolders` which is not very convenient.

I am somewhat lost now. Not sure which path i should put more effort in :D

Luegengladiator, 2021-11-20:
Yes you are right, the mpd has to load the configuration. However, this is only once or when additional folders have been added. 
I'm not aware of a preload function in the webserver. That's exactly what I'm looking for at the moment.

Thilo-90, 2021-11-20:
Not sure what you mean by "Signal comes up". 

I can swipe rfid cards no Problem and the folders/ Web radios are played without any Problem. 
I just cannot open the Webserver on my phone or Windows PC when the NAS is mounted.

Luegengladiator, 2021-11-20:
Ok perfekt. Then we are sure it's the Webserver. 
There is a Signal after Startup showing "ready for action" but maybe it's not configured by default.
I'll look at the Webservice now.

Luegengladiator, 2021-11-20:
Yes you are right. The Website does a preload of the folder structure:
inc.viewFolderTree.php
```
// get all containing folders and subfolders
$subfolders = dir_list_recursively($audiofolder);
```

and as it's on the index.php 
```
// go through all folders
foreach($audiofolders as $audiofolder) {
    include('inc.viewFolderTree.php');
}
```
you pay the price regardless of local files or NFS/CIFS.
The time is lost because of the big Media Library.

So I'm pretty sure it's not a long loading Webserver, it's the index.php taking ages on big Media Libraries.
Maybe @zxa can help here? It might be a good idea to cache this kind of data?

Thilo-90, 2021-11-20:
Thank you @Luegengladiator 
That means my suspicion was correct. So i will just see and wait if caches are possible. Would this be a feature request?

Meanwhile i will try to get the streaming URLs working or just mount smaller sub-folders.

Thilo-90, 2021-11-21:
I did not yet implement functioning streaming URLs to my NAS, but thinking about it another issue came across my mind.

I can only link single mp3 files with a streaming url.
This might be useful for long tracks, but audiobools are usually in folders with several smaller tracks.

Would it be possible to stream a folder as a whole? Or make a playlist and let it stream that playlist of mp3s?

Luegengladiator, 2021-11-21:
@Thilo-90 
That's exactly how the system is designed. If you link a RFID to a Folder, a Playlist is generated and so it is working for Audiobooks as well.

Thilo-90, 2021-11-21:
But can I link folders on my nas which is not mounted? So instead of "audiofolders/nas/audiobook1/" can I link "192.168.xxx.xxx/volume1/audiobook1/"? 

I do not have this option in the Webserver I guess.

Luegengladiator, 2021-11-21:
sorry I misunderstood that. 
This might be possible, if you are using URLs to the Stream. Same as a Online-Radio stream. 
But then it's a problem with Audiobooks. 

I'm thinking of "automounter" instead to mount the directory only when it's needed. But then, to be honest, the WebUI will pay the additional price. 

thinking about it, we might get an faster way to the UI Information we need. 
@s-martin @pabera @zxa @arne123 @MiczFlor is it really needed to read the hole filesystem structure in the UI? 
mpc is able to list the structure really fast because of the underlaying DB:
to get the root directory structure
`mpc ls /`
or to get a dedicated folder without leading / 
`mpc ls "Audioboos/Path to the folder"`

pabera, 2021-11-25:
> is it really needed to read the hole filesystem structure in the UI?

We don't do this in our [future3 release](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-) anymore. We only use MPD DB commands. Hence the performance should be much better although a stress-test has not been run yet. I don't have a lot of experience how MPD works with mounted network devices (specifically when they are unmounted). But it shouldn't be a big deal

Luegengladiator, 2021-11-25:
@pabera thank you for clarification and good to hear this is not an issue any more!

mpd works on the local database until rescan. 
So if the device is not accessible there will be an error like "file not found".
I'll recheck the MPD config if there is something we have to take care of.

Luegengladiator, 2021-12-28:
I checked my config. 
I have pinned some points like bind_address and port but there is not critical change in my MPD conf. 
@Thilo-90 as @pabera already mentioned this is solved in future3 can we close this issue?

Thilo-90, 2021-12-31:
Yes can be closed. Will check the future 3 Release and meanwhile just use a big sd card :) 

Thanks all for your help!

closed 2021-12-31 by Thilo-90