MiczFlor/RPi-Jukebox-RFID #1633: Webapp / websockets not working

weltraumkuh, 2021-11-20:
The webapp is not working/communicationg with the Jukebox daemon.

I can controle the Jukebox via comandline "run_rpc_tool.py" (start stop playback get list of files etc.) get status via "run_publicity_sniffer.py" including  card swipes. 
But the webapp don't show any status, filelist or cards and does not allow control.
The issue seams to be the websocktes, as firefox shows error the message "Firefox kann keine Verbindung zu dem Server unter ws://192.168.66.150:5556/ aufbauen." (can not connect to server ...).

Netstats show that python is listening on the ports:
pi@raspberrypi-z2w:~/RPi-Jukebox-RFID $ netstat -t -l -p -n
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
…
tcp        0      0 0.0.0.0:5555            0.0.0.0:*               LISTEN      16025/python3       
tcp        0      0 0.0.0.0:5556            0.0.0.0:*               LISTEN      16025/python3       
tcp        0      0 0.0.0.0:5557            0.0.0.0:*               LISTEN      16025/python3       
tcp        0      0 0.0.0.0:5558            0.0.0.0:*               LISTEN      16025/python3       
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
…

Accessing from localhost via SSH portforwarding does not change anything.

The rpc server seams running and does not output any error only a successfull start:
20.11.2021 23:09:31 -   67:server.py          - jb.rpc.server        - MainThread      - INFO     - Init RPC Server (Pyzmq version: 22.3.0; ZMQ version: 4.3.4; has draft API: False)
20.11.2021 23:09:31 -   75:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Connected to address 'inproc://JukeBoxRpcServer'
20.11.2021 23:09:31 -  100:multitimer.py      - jb.multitimers       - host.timer.cputemp - DEBUG    - Start timer 'host.timer.cputemp in endless mode
20.11.2021 23:09:31 -   81:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Connected to address 'tcp://*:5555'
20.11.2021 23:09:31 -   87:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Connected to address 'ws://*:5556'
20.11.2021 23:09:31 -   92:server.py          - jb.rpc.server        - MainThread      - INFO     - All socket connections initialized
20.11.2021 23:09:31 -  219:daemon.py          - jb.daemon            - MainThread      - INFO     - Start-up time: 1821.827 ms
20.11.2021 23:09:31 -  104:server.py          - jb.rpc.server        - MainThread      - INFO     - RPC Servers started

My hardware only for completeness:
RPi Zero 2 w
RC522 SPI Reader
Miniamp

weltraumkuh, 2021-11-21:
Finally I did the obvious and tried an other browser, it is working with Chrome and Safari, my Firefox Versions on Mac (94.0.1) and Android (94.1.2) seam to have issues with zeroqm.

closed 2021-11-21 by weltraumkuh

pabera, 2021-11-21:
Correct.. jszmq does have an issue on Firefox. We haven't bothered to fix this issue until now but have to look into it once we look at a stable release.

andreasbrett, 2021-11-28:
> Correct.. jszmq does have an issue on Firefox. We haven't bothered to fix this issue until now but have to look into it once we look at a stable release.

Might be worth a mention in https://github.com/pabera/RPi-Jukebox-RFID/blob/b8a6ec56ac90410345025d2325976992a2f29a02/docs/development/docker.md.

I wasted some time today getting a dev environment up and running. First finding this docker documentation (btw: in which branch is that? could only find it linked in some issue IIRC. would be better linked in the wiki or [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/CONTRIBUTING.md)). Then trying to find out why the webapp is not working and finally arriving here.

weltraumkuh, 2021-12-07:
thanks a lot, that would have spared me a lot of trouble.