MiczFlor/RPi-Jukebox-RFID #1636: 🚀 | Assign RFID directly to track

zimbo86, 2021-11-22:
## Assign RFID directly to track
### What functionality would you like to see in your phoniebox?

I want to assign an RFID chip directly to a track, not to a complete directory

### How do you envision the feature to work from a users perspective?

The idea is a set of animal sounds. RFID chips are tagged to animal toys. Actually (I think) I have to add a folder for every anmimal
DOG\dog.mp3
CAT\cat.mp3
etc.
and use the "single" option.

I´m looking for a way to create one folder and assign a chip directly to an audio file in ist
ANIMALS\cat.mp3
ANIMALS\dog.mp3

pabera, 2021-11-25:
We will be able to release this feature with the [future3 release](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-). I already have it running locally and I will merge it in the next couple of days.

s-martin, 2023-05-05:
@pabera, now available on future3?

s-martin, 2023-12-17:
Available for version 3.x (future3)

closed 2023-12-17 by s-martin