MiczFlor/RPi-Jukebox-RFID #1644: 🐛 | BUG SUMMARY

6RomN6, 2021-11-26:
## Bug

### What I did

### What happened

### I expected this to happen

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Plus Rev 1.3

### RFID Reader

RFID-RC522

### Other notable hardware

XTNs6QTTm5hf, 2021-11-29:
Hi,

problem is related to slightly different behaviour of raspbian bullseye compared to officially supported raspbian buster. I think I had the same issues when trying to install on a new pi zero 2 using bullseye. 

There are two problems in buster install script:
1) Buster install script has hard coded references to php 7.3 while bullseye is installing php 7.4
2) Bullseye intsall script fails with some strange gcc errors you where posting in your bug report. This can be fixed by using special cflags in "pip install" lines of the script.

In issue #1623 you can find an updated install script working on a pi zero2 running bullseye: # https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1623#issuecomment-968364964

6RomN6, 2021-11-29:
I was able to do the install with this script without the errors before... but the rfid reader still not work...i have done the stuff they do in #1623 but it haven't help... but if i have understand it right, this things aren't nessesary?

olsins, 2021-12-01:
do the installation skript - when you are asked about the RFID reader press n
after reboot try 
sudo python3 /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py

select the device - for me the script started to "hang" after the selectio --> just press STRG + C
you are good to go

XTNs6QTTm5hf, 2021-12-02:
> I was able to do the install with this script without the errors before... but the rfid reader still not work...i have done the stuff they do in #1623 but it haven't help... but if i have understand it right, this things aren't nessesary?

I usually forget to activate SPI interface in raspi-config (Interfacing Options, Enable SPI). Without the RFID reader wont work

6RomN6, 2021-12-02:
both thing i have tried...
i have found somewhere a command to show all activ spi devices... sadly i can't find the command any more... this is why i think the wiring is correct... because i have seen the reader...

pabera, 2021-12-02:
Would [Version 3](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-) be an option for you instead? It has better RFID reader support.

V2.X and Pi OS Bullseye have not been tested properly because Bullseye is rather new. If not already the case, you might want to downgrade to Pi OS Buster and try it again.

6RomN6, 2021-12-02:
I have tried both ways with no luck. Version 3 if i remeber correct... the installation stops somewhere...and with the downgrade i get the same error like i had when i opened the issue... but i'm not sure i remeber correct...

just to be sure that i have make it right... where and which image would download for a downgrade?

i will first try a downgrade... if this not work i would try the Version 3

pabera, 2021-12-02:
You can download Buster Lite from here: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/

Use the Pi Imager and install it as a custom image. The default image installation will use Bullseye

> Version 3 if i remeber correct... the installation stops somewhere.

It does get better everyday, so might want to give it a try :-)

6RomN6, 2021-12-05:
I think it work now, with the downgrade... i haven't tried the V3 because i will need the spotify... and as far i can see there is no spotify yet...

Thanks for your help!

pabera, 2021-12-05:
I am glad to hear it works.

And you're right. Spotify is still on our roadmap for V3.

closed 2021-12-05 by pabera

grypho, 2021-12-22:
This issue is related to GCC10. Raspios changed the default compiler from GCC9 to GCC10 which results in a different behaviour when compiling the Python modules. When you add the compiler directive -fcommon to all PIP calls, the install script runs without any issues. 

Example:

` sudo CFLAGS="-fcommon" python3 -m pip install --upgrade --force-reinstall -q -r "${jukebox_dir}"/requirements.txt`