MiczFlor/RPi-Jukebox-RFID #1653: Fehlermeldung nach Phoniebox-Installation

Borlizgaron, 2021-12-01:
##An error occured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerpause -v=single Output: RC: .1

### Hallo zusammen,
ich bin etwas verzweifelt, installiere ich doch meine Phoniebox zum xten mal. Am Ende, nach Auswahl des RFID-Readers, kommt eine Fehlermeldung.
Ich kann zwar in die Oberfläche der Phoniebox, und per Karte regiert der RFID-Reader, allerdings kann ich weder Songs hochladen noch wird die Nummer der Karte angesagt. Unter Settings ist der RFID-Reader disabled und lässt sich auch nicht nach einem "Switch ON" und einem "Submit" immer noch disabled.
Geht man in den Player und drückt einmal auf einen Button der Oberfläche, dann erscheint diese Fehlermeldung:
An error occured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerpause -v=single Output: RC: .1

Mein Setup:
- Pi 3B
- GPIO Erweiterung Extension Board
- Hifiberry MiniAmp (Soundtest vor Installation durchgeführt - funktioniert)
- On/Off-Shim

Ich habe den Installationsweg von splitti durchgeführt: 
https://splittscheid.de/phoniebox-bauanleitung-toniebox-alternative/#ftoc-installation-software

### Und hier die Fehlermeldung während der Installation:

> Generating locales (this might take a while)...
>   en_GB.ISO-8859-1... done
>   en_GB.UTF-8... done
> Generation complete.
> Created symlink /etc/systemd/system/multi-user.target.wants/phoniebox-gpio-control.service → /etc/systemd/system/phoniebox-gpio-control.service.
> Updating DB (#2) ...
> volume: n/a   repeat: off   random: off   single: off   consume: off
> Setting /etc/dhcpcd.conf...
> Setting /etc/wpa_supplicant/wpa_supplicant.conf...
> Starting dhcpcd service...
> Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
> Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
> perl: warning: Setting locale failed.
> perl: warning: Please check that your locale settings:
>         LANGUAGE = "de_DE.UTF-8",
>         LC_ALL = "de_DE.UTF-8",
>         LANG = "de_DE.UTF-8"
>     are supported and installed on your system.
> perl: warning: Falling back to the standard locale ("C").
> perl: warning: Setting locale failed.
> perl: warning: Please check that your locale settings:
>         LANGUAGE = "de_DE.UTF-8",
>         LC_ALL = "de_DE.UTF-8",
>         LANG = "de_DE.UTF-8"
>     are supported and installed on your system.
> perl: warning: Falling back to the standard locale ("C").
> Setting owner and permissions for directories...
> amixer: Cannot find the given element from control default
> 
> 
> #
> # INSTALLATION FINISHED
> 
> 
> 
> #####################################################
> 
> If you are using an RFID reader, connect it to your RPi.
> (In case your RFID reader required soldering, consult the manual.)
> Have you connected your RFID reader? [Y/n] y
> Please select the RFID reader you want to use
> 1) USB-Reader (e.g. Neuftech)  4) Manual configuration
> 2) RC522                       5) Multiple RFID reader
> 3) PN532
> #? 1
> Traceback (most recent call last):
>   File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
>     from Reader import get_devices
>   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
>     from evdev import InputDevice, ecodes, list_devices
> ModuleNotFoundError: No module named 'evdev'
> chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
> chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
> 
> DONE. Let the sounds begin.

Ich glaube nicht, dass der RFID-Reader kaputt ist. Auch habe ich es einmal versucht mit aktueller Raspian-Version und einer älteren (also ohne Update).
Ich habe die Verkabelung der GPios nach der Folgeinstallation nicht wieder abgebaut, könnte das zu Problemen führen? Oder muss der On/Off-Shim erst später installiert werden?

pabera, 2021-12-02:
Anhand deiner Beschreibung konnte ich nicht ganz herausfinden, ob du Bullseye (neues Pi OS, kommt mit Pi Imager 1.6.2) oder Buster (Pi Manager 1.6) als Basis OS benutzt. Version 2.x wurde für Buster gebaut und getestet, vor kurzem wurde aber Bullseye released und jene Version wurde noch nicht auf Herz und Nieren getestet. Vielleicht versucht du es mit Buster noch einmal.

Oder versuchst [Version 3](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-), die funktioniert schon recht stabil, ist aber noch in der Entwicklung.

juduroe, 2021-12-03:
Ich hatte gerade ebenfalls Probleme mit meinem RFID Reader (auch RC522).
Was zeigt bei dir `systemctl status phoniebox-rfid-reader.service`? Wenn hier ein Fehler auftaucht, kannst du mal schauen, was `journalctl -r` ausgibt. Dort wurde bei mir mit der folgenden Meldung auf fehlende python3-Module hingewiesen:

> Dez 03 17:50:12 phoniebox daemon_rfid_reader.py[2218]: ModuleNotFoundError: No module named 'pirc522'
> Dez 03 17:50:12 phoniebox daemon_rfid_reader.py[2218]:     import pirc522

Bei mir haben insgesamt zwei python3-Module gefehlt (evdev und pi-rc522). Nachdem ich diese mit `sudo pip3 install evdev pi-rc522` nachinstalliert habe, konnte ich mit `sudo systemctl start phoniebox-rfid-reader.service` den RFID-Dienst wieder starten. Anschließen hat auch das Eintragen neuer RFID-Tags im Webinterface funktioniert.

@dev: sollte hier die requirements.txt erweitert werden?

Borlizgaron, 2021-12-03:
> Anhand deiner Beschreibung konnte ich nicht ganz herausfinden, ob du Bullseye (neues Pi OS, kommt mit Pi Imager 1.6.2) oder Buster (Pi Manager 1.6) als Basis OS benutzt. Version 2.x wurde für Buster [...]

Ha! Vielen Dank. Das war das Problem! Ich hatte Bullseye genutzt. Mit Buster funktioniert alles ohne Fehlermeldung. Vielen Dank!

closed 2022-03-24 by s-martin