MiczFlor/RPi-Jukebox-RFID #1654: Usage of GPIO7 in gpio_settings.ini crashes phoniebox-gpio-control

Caliandroid, 2021-12-01:
## Bug
Reinstalled Phoniebox on a RasperryPi3B with Raspi OS (bullseye) and afterwardsRaspbian Buster some days ago with most recent install script: Everytime the GPIO buttons did not work.

I isolated the error to the magic GPIO 7 (i can even switch the GPIO on the hardware to avoid it).
As soon as i use 7 in the gpio_settings.ini (see below),   phoniebox-gpio-control services crashes, as soon as I change it against 6 or another digit <10, the gpio service runs correctly. 
I have a 2nd Raspi3b with the same GPIO 7 config (as i have two kids) and the other one works just fine.

Workaround: I've changed the cables from GPIO 7 to 15 = no more problems.
Could it be a hardware problem or does anyone encounter the same issue?

```
pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ sudo systemctl status  phoniebox-gpio-control
● phoniebox-gpio-control.service - Phoniebox GPIO Control Service
   Loaded: loaded (/etc/systemd/system/phoniebox-gpio-control.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2021-12-01 20:41:38 GMT; 1s ago
  Process: 7063 ExecStart=/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py (code=exited, status=1/FAILURE)
 Main PID: 7063 (code=exited, status=1/FAILURE)
pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ 
```

This is my gpio_settings.ini, you'll find the 7 under [PrevNextControl]

```
[DEFAULT]
enabled : True

[VolumeControl]
enabled : True
type : TwoButtonControl
hold_time : 0.3
timebase : 0.1
functioncalltwobuttons : functionCallVol0
pull_up_down : pull_up
hold_mode : Repeat
pin1 : 13
pin2 : 5
functioncall1 : functionCallVolU
functioncall2 : functionCallVolD

[PrevNextControl]
enabled : True
type : TwoButtonControl
pin1 : 10
pin2 : 7
functioncall1 : functionCallPlayerPrev
functioncall2 : functionCallPlayerNext
functioncalltwobuttons : None
hold_time : 0.3
pull_up_down : pull_up
hold_mode : None

[PlayPause]
enabled : True
type : Button
pin : 21
hold_time : 0.3
functioncall : functionCallPlayerPause
pull_up_down : pull_up

[Shutdown]
enabled : True
type : Button
pin : 27
hold_time : 2
functioncall : functionCallShutdown
pull_up_down : pull_up

[Volume0]
enabled : False
type : Button
pin : 17
hold_time : 0.3
functioncall : functionCallVol0
pull_up_down : pull_up

[VolumeUp]
enabled : False
type : Button
pin : 16
hold_time : 0.3
functioncall : functionCallVolU
pull_up_down : pull_up
hold_mode : Repeat

[VolumeDown]
enabled : False
type : Button
pin : 19
hold_time : 0.3
functioncall : functionCallVolD
pull_up_down : pull_up
hold_mode : Repeat

[NextSong]
enabled : False
type : Button
pin : 26
hold_time : 0.3
functioncall : functionCallPlayerNext
pull_up_down : pull_up

[PrevSong]
enabled : False
type : Button
pin : 20
hold_time : 0.3
functioncall : functionCallPlayerPrev
pull_up_down : pull_up

[Halt]
enabled : False
type : Button
pin : 21
hold_time : 0.3
functioncall : functionCallPlayerPauseForce
pull_up_down : pull_up
```

### Further information that might help

## Software

### Base image and version
Raspbian GNU/Linux 10 (buster)"
Raspbian GNU/Linux"
VERSION_ID="10"

but had it before with the recent bullseye release too.

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2022-12-27:
Can you check, if the issue still persists with current 2.4 version?

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin