MiczFlor/RPi-Jukebox-RFID #1659: 🐛 | Spotify-Musik startet nicht - Problem mit Mopidy

Borlizgaron, 2021-12-04:
##Mopidy: Failed to add some tracks

### Hallo zusammen,
Spodify-Playlists oder Alben lassen sich nicht über Iris und damit auch nicht über die Phoniebox abspielen. In Iris habe ich entdeckt, dass beim Anspielen dann die Fehlermeldung "Mopidy: Failed to add some tracks" erscheint. Ich kann aber problemlos durch Iris navigieren, finde meine Favoriten, Playlists und Alben. 
Die Box funktioniert sonst tadellos. Lokale Musik wird gespielt, RFID-Karten gehen und alle Knöpfe funktionieren.

Mein Setup:
- Pi 3B
- Raspian Buster Full
- GPIO Erweiterung Extension Board
- Hifiberry MiniAmp
- On/Off-Shim

Ich habe den Installationsweg von splitti durchgeführt: 
https://splittscheid.de/phoniebox-bauanleitung-toniebox-alternative/#ftoc-installation-software

Dabei wird bei dem Punkt der Phonieboxinstallation die Einrichtung von MPD abgelehnt. Das sollte aber nichts damit zu tun haben, oder?

Ich habe mich durch die FAQ durchgearbeitet, aber nichts scheint zu helfen.
Ich habe jetzt, damit die Box etwas schneller ist in der mopidy.conf 
allow_playlists = false auskommentiert. Aber selbst, wenn ich das rückgängig mache, hat das keine Auswirkungen.

Hat jemand vielleicht eine Lösung oder Idee?

UbiGIT89, 2021-12-04:
Hi all,

same problem here.

I can reproduce the issue wirh many albums of "Conny" (german kids stories).
Additional to this, albums which are linked to rfid cards can be played by card without any issue.
Same album directly via iris/spotify cannot.

The logs says only "Cannot add some tracks" but no reason.

Best regards
Torben

FranzMan, 2021-12-05:
I have the same problem (at least I think so)

Additionally if it try to play an RFID-Card it only seems to load the content, but it doesn't start to play.
Also if I click the big play-button of a listed item below the player it does the same (only loading), but if I click on the little play-button of a file in the playlist, it does play this mp3.

FranzMan, 2021-12-05:
Nachdem Ich es jetzt nochmal mit der Lite-Variante von Raspberry probiert habe geht es. Nur Ordner abspielen geht nicht, aber Spotify schon und einezelne Dateien auch.
Man muss dann halt alles via ssh und putty machen.

Borlizgaron, 2021-12-11:
I have now installed the Lite version and there is no difference. In Iris I see that the playlist is playing, but after 3 seconds the Iris player aborts and repeats. Nothing comes out of the Phoniebox. "Mopidy: Failed to add some tracks".

--------

Ich habe nun die Lite-Version aufgespielt und es gibt keinen Unterschied. In Iris sehe ich, dass die Playlist abgespielt wird, aber nach 3 Sekunden bricht der Iris-Player ab und wiederholt. Aus der Phoniebox kommt nichts. "Mopidy: Failed to add some tracks"

matu1504, 2021-12-25:
Hatte das Problem auch, das Problem lag bei mir am Systemstart.
Mopidy wird gestartet und das WLAN ist noch nicht connected,
Probier mal in der Konsole folgendes

`sudo` /etc/init.d/mopidy restart

bei mir ging es dann.
Ich habe nun den Start von Mopidy deaktiviert und starte es durch ein Script 30 Sekunden nach dem Bootvorgang.
Damit geht es.

Geht sicher auch einfacher, aber mit den Abhängkeiten in den init.d Script bin ich nicht  so vertraut.

Borlizgaron, 2021-12-25:
Oh, super. Kannst Du mir das Skript erläutern? Es wird im Bootvorgang ausgeführt? Hast Du es in der Config eingefügt?
Es macht aber den Startvorgang nicht langsamer?

matu1504, 2021-12-25:
Wenn der Restart des Mopidy dein Problem gelöst hat, dann schreib ich dir mal was ich gemacht habe.

Vielleicht keine schöne Lösung, aber für mich funktioniert es.
Erst habe ich den mopidy Service beim Systemstart deaktiviert

`sudo update-rc.d mopidy disable`

Dann hab ich im root Ordner eine Datei mit dem Name delay angelegt
`sudo nano ~/delay`

folgender Inhalt:
`#!/bin/bash`
`# $1: delay time(s)`
`# $2: command`

`sleep $1`
`exec $2`

anschließend ausführbar machen
`sudo  chmod a+x ~/delay`

cronjob einrichten, der 30 Sekunden nach Bootvorgang den mopidy startet
`sudo crontab -e`

Der sieht so aus:
`@reboot /root/delay 30 "/etc/init.d/mopidy start"`

fertig.

Schön wäre es ihn nur für die RCs für den Start zu deaktivieren und nicht für das runterfahren, aber das habe ich aus Einfachheit nicht gemacht.

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin