MiczFlor/RPi-Jukebox-RFID #1666: 🚀One Button for Next Song and Skip Forward

Nacktmull87, 2021-12-07:
Hi everyone,

is it possible to give one Button  two functions of Play Next Song when Pushed The Button for a short Time and skip Forward when its hold for a View Seconds?

I tried IT, but when i hold The Button its jumping to the Next Song and then IT Starts to skip in the song. Is there Any Chance that this will not Happen? 

I Hope you will understand what i mean.

Thx for your help.

Sry for my Bad english

TheDuffman85, 2023-09-08:
Only thing that worked for me is to reverse the order. Forward seek on short push and next title on long push. If you long push the button first it will perform a forward seek and then skip to the next title.

AlvinSchiller, 2024-05-05:
Referenced Tickets:
https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1999
https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2247
https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2365

https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1827

AlvinSchiller, 2024-05-08:
This is now fixed with #https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2367 in v2.8.0.
The behavior of `hold_mode` `SecondFunc` and `SecondFuncRepeat` has changed. The secondary function is no longer triggered additionally to the primary function.
Now its called exclusively if `hold_time` is reached. The primary function will only be triggered if the button is pressed shorter then `hold_time`! Existing configurations may need to adapt to this.

closed 2024-05-08 by AlvinSchiller