MiczFlor/RPi-Jukebox-RFID #1667: Attempted import error: 'Redirect' is not exported from 'react-router-dom'

jeverling, 2021-12-08:
Hi, I'm trying to rebuild the webapp, and it's giving me the following error: 

```
Creating an optimized production build...
Failed to compile.

./src/components/Library/lists/index.js
Attempted import error: 'Redirect' is not exported from 'react-router-dom'.
```

I'm on the latest commit of `future3/develop` - `a12746f7565823d264c7511df8a1853a691af58a`

The problem seems to be an update in _react-router-dom_: https://stackoverflow.com/a/63691192
Suggested solution is to replace `<Redirect to='/' />` with `<Navigate replace to="/" />`, or pin an older version of _react-router-dom_.

pabera, 2021-12-08:
Hey, thanks for checking. I have a new [PR](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1665) where this will be fixed. I will merge it tomorrow most likely.. A `git pull` and a rebuild will do the trick then..

pabera, 2021-12-09:
Problem should be solved

```
cd; ENABLE_WEBAPP_PROD_DOWNLOAD=true GIT_USER='MiczFlor' GIT_BRANCH='future3/develop' bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)
```

closed 2021-12-09 by pabera

jeverling, 2021-12-09:
Can confirm it is resolved (I updated `future3/develop` branch and then ran `./run_rebuild.sh -u` in `src/webapp` folder).
Thanks a lot! :)