MiczFlor/RPi-Jukebox-RFID #1670: 🐛 | Web interface does not always start - lighttpd fails to start

alexanderkoller, 2021-12-11:
## Bug

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script.

### What happened

Sometimes the Phoniebox does not start its web interface. Sometimes it does, so this is not a duplicate of any of the issues I could find; but of course the web interface should come up reliably.

### I expected this to happen

That the web interface starts reliably.

### Further information that might help

Here's what syslog has to say about lighttpd:

```
pi@raspberrypi:/var/log $ grep lighttpd syslog.1
Dec 10 07:43:43 raspberrypi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 10 07:43:43 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Dec 10 07:43:43 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Dec 10 07:43:43 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 1.
Dec 10 07:43:45 raspberrypi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 10 07:43:45 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Dec 10 07:43:45 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Dec 10 07:43:45 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 2.
Dec 10 07:43:46 raspberrypi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 10 07:43:46 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Dec 10 07:43:46 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Dec 10 07:43:46 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 3.
Dec 10 07:43:47 raspberrypi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 10 07:43:47 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Dec 10 07:43:48 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Dec 10 07:43:48 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 4.
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Start request repeated too quickly.
Dec 10 07:43:49 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
```

I'd be happy to debug this on my own, but this is a spectacularly uninformative error message. Do you have any suggestions for what I should be looking at?

## Software

### Base image and version

```
pi@raspberrypi:/var/log $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Branch / Release

```
pi@raspberrypi:/var/log $ cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
On branch master
Your branch is up to date with 'origin/master'.
```

### Installscript

`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

```
pi@raspberrypi:~/RPi-Jukebox-RFID $ sudo cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero 2 Rev 1.0
```

### RFID Reader

AZ-Delivery RC522

### Soundcard

Hifiberry MiniAmp

### Other notable hardware

I attached a Neopixel LED ring, but can't see how this would be relevant. The Pi never reports undervoltage as far as I can tell.

alexanderkoller, 2021-12-11:
After more research, I discovered `/var/log/lighttpd/error.log`, which says

```
2021-12-11 16:28:39: (gw_backend.c.507) bind failed for: unix:/var/run/lighttpd/php.socket-0 No such file or directory 
2021-12-11 16:28:39: (gw_backend.c.1510) [ERROR]: spawning gw failed. 
2021-12-11 16:28:39: (server.c.1472) Configuration of plugins failed. Going down. 
```

Apparently this is caused by `/var/run` not being writable for `www-data`, which means that the lighttpd service can't create the `lighttpd` subdirectory (as user `www-data`).

I couldn't figure out how to permanently give write access to `/var/run` to `www-data` (it is a tmpfs), so I worked around it as follows. In `/etc/lighttpd/conf-enabled/15-fastcgi-php.conf`, I changed the filename for the PHP socket:

```
"socket" => "/tmp/lighttpd/php.socket",
```

In `/etc/systemd/system/multi-user.target.wants/lighttpd.service`, I added two lines in the `[Service]` section:

```
ExecStartPre=mkdir /tmp/lighttpd
ExecStartPre=chown www-data /tmp/lighttpd
```

This workaround seems to work for me, but perhaps there is a more permanent way to fix the permissions?

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin