MiczFlor/RPi-Jukebox-RFID #1683: Probleme mit mdp.timer_status on future3

jmber, 2021-12-21:
### Describe your problem 
Wenn eine Playlist oder Ordner nur einen Song  hat und dieser endet, kommt unten genannte Fehlermeldung. Danach ist der mpd nicht mehr nutzbar -> keine Musik mehr abspielbar, Playlists leer, etc...

src/jukebox/components/playermpd/__init__.py
Line: 193
  self.status_thread = multitimer.GenericEndlessTimerClass('mpd.timer_status',
                                                        self.mpd_status_poll_interval, self._mpd_status_poll)
       self.status_thread.start()

#### What's your hardware set up? 

Version: 3.1.0, Raspi B+, debian bulleseye

ERROR LOG:
`21.12.2021 10:32:32 -  692:plugs.py           - jb.plugin.call       - MainThread      - DEBUG    - Calling: player.ctrl.seek(args=(), kwargs={'new_time': '181.977'})
21.12.2021 10:32:32 -  154:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Sending response: {'result': None, 'id': 'eaf599ab-cdc2-4f0a-9a9d-bb9ce33a42bf'}
21.12.2021 10:32:41 -  236:__init__.py        - jb.PlayerMPD         - mpd.timer_status - ERROR    - timeout: timed out
21.12.2021 10:32:45 -  146:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '0001889390'
21.12.2021 10:32:45 -  692:plugs.py           - jb.plugin.call       - read_00Thread   - DEBUG    - Calling: player.ctrl.play_single(args=['file .mp3
21.12.2021 10:32:45 -  782:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Ignoring failed call: 'player.ctrl.play_single(args=['file .mp3
21.12.2021 10:32:45 -  783:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Reason: OSError: cannot read from timed out object
21.12.2021 10:32:45 -  784:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 776, in call_ignore_errors
    result = _call(package, plugin, method, args=args, kwargs=kwargs, as_thread=as_thread, thread_name=thread_name)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 724, in _call
    return func(*args, **kwargs)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/playermpd/__init__.py", line 404, in play_single
    self.mpd_client.clear()
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 469, in mpd_command
    return wrapper(self, name, args, callback)
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 532, in _execute
    return retval()
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 454, in command_callback
    res = function(self, self._read_lines())
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 387, in _parse_nothing
    for line in lines:
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 586, in _read_lines
    line = self._read_line()
  File "/home/pi/.local/lib/python3.9/site-packages/mpd/base.py", line 568, in _read_line
    line = self._rbfile.readline().decode("utf-8")
  File "/usr/lib/python3.9/socket.py", line 701, in readinto
    raise OSError("cannot read from timed out object")
OSError: cannot read from timed out object

21.12.2021 10:32:49 -  146:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '0001889390'
21.12.2021 10:32:49 -  692:plugs.py           - jb.plugin.call       - read_00Thread   - DEBUG    - Calling: player.ctrl.play_single(args=['file .mp3
21.12.2021 10:32:49 -  782:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Ignoring failed call: 'player.ctrl.play_single(args=['file.mp3
21.12.2021 10:32:49 -  783:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Reason: OSError: cannot read from timed out object
21.12.2021 10:32:49 -  784:plugs.py           - jb.plugin            - read_00Thread   - ERROR    - Detailed reason:
Traceback (most recent call last):
`

pabera, 2021-12-21:
Danke für die Fehlermeldung. Wir gucken uns das demnächst mal an. Welches Release hast du getestet?

jmber, 2021-12-21:
3.1.0
die aktuelle future3/develop branch

jeverling, 2021-12-28:
Ich wollte nur bestätigen, dass ich die gleiche Fehlermeldung auch gesehen habe. Ich hatte leider keine Zeit zu debuggen (Weihnachten), bin auf `e46ee20` zurück weil dieser commit bei mir gut funktioniert hat. Mit 3.1.0 bekam ich den gleichen Traceback angezeigt.

ChisSoc, 2021-12-31:
Just to give you an intermediate update: I can re-create the issue in my test setup. Curios, as I specifically wrote some code to circumvent this situation. I am on it.

ChisSoc, 2022-01-01:
I isolated the potential issue to a timeout in the mpd player interface. It would be a great help, if you could verify the fix, too:

Please change this line 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/6b72a21f5c5e5906ded14eb87a79b763d6a5f2a9/src/jukebox/components/playermpd/__init__.py#L156

into
~~~
self.mpd_client.timeout = None
~~~
A PR with the fix will follow, but independent verification would be great (if you like to do it)

jeverling, 2022-01-01:
I can confirm that this fixes the issue for me, with a fresh install (6b72a21).

ChisSoc, 2022-01-02:
> I can confirm that this fixes the issue for me, with a fresh install ([6b72a21](https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/6b72a21f5c5e5906ded14eb87a79b763d6a5f2a9)).

@jeverling: Many thanks!

ChisSoc, 2022-01-04:
Fixed in [Release 3.1.1](https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/V3.1.1)

Please update :-)

closed 2022-01-04 by ChisSoc

jmber, 2022-01-05:
@ChisSoc Genial! :-) Das ging ja fix!

PR #1701 (2022-01-02): Prevent socket timeout in MPD player interface
Fixes #1683