MiczFlor/RPi-Jukebox-RFID #1684: MQTT didnt work correctly

florianru1984, 2021-12-22:
I activated MQTT on my Phoniebox but not all topics will work.

"sudo python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py
Connecting to 192.168.178.51 on port 1883
   [LOG] Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'phoniebox'
Subscribing to phoniebox/cmd/#
   [LOG] Sending SUBSCRIBE (d0, m1) [(b'phoniebox/cmd/#', 0)]
Subscribing to phoniebox/get/#
   [LOG] Sending SUBSCRIBE (d0, m2) [(b'phoniebox/get/#', 0)]
   [LOG] Received CONNACK (0, 0)
Connection established.
   [LOG] Sending PUBLISH (d0, q1, r1, m3), 'b'phoniebox/state'', ... (6 bytes)
   [LOG] Sending PUBLISH (d0, q1, r1, m4), 'b'phoniebox/version'', ... (22 bytes)
   [LOG] Sending PUBLISH (d0, q1, r1, m5), 'b'phoniebox/edition'', ... (7 bytes)
   [LOG] Sending PUBLISH (d0, q1, r1, m6), 'b'phoniebox/disk_total'', ... (4 bytes)
   [LOG] Sending PUBLISH (d0, q1, r1, m7), 'b'phoniebox/disk_avail'', ... (4 bytes)
   [LOG] Received SUBACK
   [LOG] Received SUBACK
   [LOG] Received PUBACK (Mid: 3)
   [LOG] Received PUBACK (Mid: 4)
   [LOG] Received PUBACK (Mid: 5)
   [LOG] Received PUBACK (Mid: 6)
   [LOG] Received PUBACK (Mid: 7)
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 392, in <module>
    processGet("all")
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 156, in processGet
    mpd_status = fetchData()
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 330, in fetchData
    result["last_card"] = readfile(path + "/../settings/Latest_RFID")
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 191, in readfile
    with open(filepath, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_RFID'"

I run the script with debug activated, above the output.

s-martin, 2022-12-19:
Version 2.4 was released yesterday, which also includes PR #1496.

@florianru1984 you might want to check, if this fixes your issue.

s-martin, 2023-05-01:
Closing for now, feel free to reopen the issue, if the problem persists

s-martin, 2023-05-01:
Seems to be the same issue as #1848, was fixed in #2017

closed 2023-05-01 by s-martin