MiczFlor/RPi-Jukebox-RFID #1685: _jukebox_core_download_prebuild_libzmq_with_drafts installation routine copies everything in /home/pi into /usr/local

jeverling, 2021-12-23:
See https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/6b72a21f5c5e5906ded14eb87a79b763d6a5f2a9/installation/routines/setup_jukebox_core.sh#L61.

I think the script should only rsync `include` and `lib` folders, not all contents of `~`. Maybe the script should download and unpack the archive into a temporary location as well.

pabera, 2021-12-23:
Thanks a lot. I'll have a look. This should definitely be corrected!

jeverling, 2021-12-28:
I think this was a fluke, and `ZMQ_TMP_DIR` ended up being unset for some reason on my system. I did not see the same again when retrying the install. I think this can be closed, sorry about the false alarm!

ChisSoc, 2021-12-29:
> ZMQ_TMP_DIR ended up being unset for some reason on my system

Any idea how that happened? Did you do anything special during the installation?

jeverling, 2021-12-31:
I think what happened, is that the directory already existed on my system (from a previous install). The code only CDs into that directory, when `mkdir ${ZMQ_TMP_DIR}` succeeds.
When the directory already exists, `mkdir` will fail and the rest of the following line will not be executed: `cd ${HOME_PATH} && mkdir ${ZMQ_TMP_DIR} && cd ${ZMQ_TMP_DIR}`.

I guess it would be good to remove the temporary directory after that step has completed.

ChisSoc, 2021-12-31:
Good analysis! Definitely looks like something we should make more robust in the install script. I am reopening this issue. @pabera: over to you, again.

s-martin, 2023-11-18:
Closing as this should be fixed in #1749 

feel free to reopen, if the issue still occurs.

closed 2023-11-18 by s-martin