MiczFlor/RPi-Jukebox-RFID #1690: "iw: command not found" on future3

jeverling, 2021-12-28:
Hi, when trying to enable the Autohotspot feature, I get the following error on a fresh Raspbian install:

```
Install AutoHotspot functionality
/home/pi/RPi-Jukebox-RFID/installation/routines/setup_autohotspot.sh: line 5: iw: command not found
/home/pi/RPi-Jukebox-RFID/installation/routines/setup_autohotspot.sh: line 6: iw: command not found
```

The script should probably check whether iw is already installed, and try to install it if not (or just try to install it either way).

jeverling, 2021-12-28:
Ah it seems to be a permission issue, the `pi` user cannot execute `iw`.

ChisSoc, 2021-12-29:
Is this in anyway related to #1685? Did the issue persist in a re-install?

Groovylein, 2021-12-29:
Could you try executing following command with the user `pi` and provide the output:
`iw dev`

jeverling, 2021-12-31:
Hi, @ChisSoc I don't think it's related. It seems to be just a permission issue.
@Groovylein here the output of the commands, I also added `sudo iw`which works fine:

```
➜  ~ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
➜  ~ iw dev             
zsh: command not found: iw
➜  ~ sudo iw dev        
phy#4
        Interface wlan0
                ifindex 4
                wdev 0x400000001
                addr 28:f3:66:21:ab:d6
                ssid Redacted
                type managed
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 20.00 dBm

```

Groovylein, 2021-12-31:
So you are using a zsh. Could you run the `iw dev` command in bash as well?

jeverling, 2022-01-01:
> How did you start the install script? With the following command?

Yes, exactly.

You might be right, and it's zsh. 😮
I just started with a fresh installation of `2021-10-30-raspios-bullseye-armhf-lite.img`, and the `pi` user can execute `iw`.
I'll let the install script run through, check if that maybe changes anything in terms of `iw` and then install zsh to see if the behavior changes then.
But probably this will not affect anybody not running zsh (or it's something else that happened on the installation but I wouldn't know what at the moment).

jeverling, 2022-01-01:
I can confirm it's actually [ohmy]zsh. Before installation, the `pi` user can execute `iw`. After ohmyzsh installation (standard configuration), the `pi` user cannot execute `iw` anymore.
It seems to be a problem with locating the executable, calling it with absolute path still works with zsh. Using bash it also still works. I have opened a ticket with ohmyzsh.

Thank you for helping debug this, if someone has the same problem in the future hopefully they will find this issue, or even better it gets fixed on the ohmyzsh side.

closed 2022-01-01 by jeverling

jeverling, 2022-01-02:
The solution is that `iw` is located in a sbin directory, `/usr/sbin/iw`.
 bash includes the sbin directories in $PATH, zsh (or at least ohmyzsh) does not. So there is no bug, just a different $PATH.

Groovylein, 2022-01-02:
Ok thabk you for testing. I will enhance the script to use the absolute path for iw (and other commands too) so that this won't happen again even in bash 😉

ChisSoc, 2022-01-02:
I don't think it is necessary use the absolute path for ALL commands. Just for the one which causes problems here! Otherwise, we have a script which never observes $PATH variables in case somebody does something custom with their shell setup.

Also, readability is better when it's just the command. What do you think?

Groovylein, 2022-01-02:
> I don't think it is necessary use the absolute path for ALL commands. Just for the one which causes problems here! Otherwise, we have a script which never observes $PATH variables in case somebody [...]

Then I wouldn't change the `iw` command either. Since it is part of the `$PATH` of the standard bash.