MiczFlor/RPi-Jukebox-RFID #1691: 🐛 | spotify:playlist dont work (spotify:album and spotify:artist works fine)

Lippsson, 2021-12-28:
Hi Guys!
I have a setup with Version2.3 - 2425890 - master
EditionPlus edition (feat. Spotify integration)

My issue is that i could not play Spotify playlists from the Player webinterface. (Iris works fine)
Strange that when i try with spotify album(s) or Artists it works fine.
How i do:

- I open iris interface
- I copy the url from the wished album/artist/playlist
- I switch to Card iD, link to a new folder paste the Link stream und select spotify as type
Note: The title.txt are always blank and i have to open the txt to write in the folder name
Note2: cover download dont work and i manually paste in the cover.jpg

- Then i scan the music libary again.
Now artists and alums will work, but i could not get playlist att work.
(When i play the playlist from Iris it works.)
I tried to delay the mopidy like matu1504 recommend with no success for playlists playback ([link](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1659#issuecomment-1001067209))

kruegmar, 2022-01-10:
I have the same problem with spotify:playlist. 😅

YildirayJuelz, 2022-01-15:
I had the same problem. [(#1961)](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1691) (sudo /etc/init.d/mopidy restart) did not help in my case.
I checked the mopidy log file. It said something about a failed verfication at start.
A new install of the phoniebox with new generated spotify [(client id and client secret)](https://mopidy.com/ext/spotify/#authentication) did the trick for me.

Here is what I guess is the problem: At startup the playlists in the linked spotify account are scanned. Only the scanned playlists will work later. Since the verification at startup failed there are no spotify playlists available later.

kruegmar, 2022-01-15:
> I had the same problem. [#1961](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1691) (sudo /etc/init.d/mopidy restart) did not help in my case.
> I checked the mopidy log file. It said something about a failed verfication at start.
> A new installation of the phoniebox with newly generated spotify [client id and client secret](https://mopidy.com/ext/spotify/#authentication) did the trick for me.

Thanks very much! This solution works for me too. Everyone is happy now. 😃

Lippsson, 2022-01-17:
Thanks @YildirayJuelz for the update.
Do i understand right that it need a reinstall to scan and implement all playlists (new) playlists
or does ist never words for playlists who are created after the install?
Will test next week with a install.

YildirayJuelz, 2022-01-17:
No, I meant somehing else. It's not about the playlist's time of creation. The playlists are scanned at every reboot.
But rebooting the raspberry pi or restarting mopidy did not help for my problem with the playlists.

From what I understand you have the same problem that I had and playlists don't work at all (?)
At least that was the case with my phoniebox.

There was something wrong with my spotify login credentials. So I created new ones and made a fresh installation with the new credentials. (The client id and client secret generated at the [mopidy website](https://mopidy.com/ext/spotify/#authentication)  worked. My old credentials were created [at the spotify dashboard](https://developer.spotify.com/dashboard/applications). Maybe that was a problem, I don't know.)

Lippsson, 2022-01-20:
YES, FINALY. I after i generated new spotify client id and client secret i placed them in /etc/mopidy/mopidy.conf
Now its fine, exept the little detail that i get no namnes when i paste the playlists in the phoniebox, but this are details and i can fix it manually.

closed 2022-01-20 by Lippsson

YildirayJuelz, 2022-01-20:
Maybe [this](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1309#issuecomment-948510269) will help with your new problem.