MiczFlor/RPi-Jukebox-RFID #1698: Second swipe defect after playlist runs out

ChisSoc, 2022-01-01:
### Describe your problem 

Second swipe does not decode to first swipe after playlist has runs out. This means that the same playlist cannot be started again when Second Swipe is enabled w/o playing something else in between

lenlennart, 2022-01-19:
see #1520 . There is also a pending PR for 2.x release

ChisSoc, 2022-01-20:
@lenlennart: Thanks for linking to the issue and committing a PR with a fix. Version 2 and Version 3 have an entirely different code base and I am an maintainer only for Version 3 - so cannot merge your PR. Still an important issue also for Version 3: it's a situation that I did not have in mind, yet. I will check it.

lenlennart, 2022-01-21:
still, the source of the problem might be the same: mpc beeing send to "pause" on removal of card, regardless of current playing state -> you should check for status = "play".

ChisSoc, 2022-01-21:
You are right there! Thanks for making the connection.