MiczFlor/RPi-Jukebox-RFID #1706: Enable start of MPD player interface even w/o MPD running

ChisSoc, 2022-01-04:
Jukebox should start w/o errors even if MPD connection cannot be estabilshed at the time of startup. Implementation is simple by test re-connecting to MPD when an MPD command is issued.

This could allow to drop the mpd.service requirement from the jukebox.service configuration and allow to start the Jukebox process in parallel to MPD and gain boot up speed.

Based on idea from the discussion in #1703.

ThomasRausU, 2022-01-05:
yes, I think this should tried. You aware that the MPD service will be started by the first access to the mpd.socket!? So there should be no dependency ("Requires or After") in the jukebox-daemon.service

ChisSoc, 2022-01-05:
Yes, I know. However, starting MPD takes times. Uur current code does not wait for MPD to start when it access the socket. We need to code that into our MPD interface. Only after that, we can change the service configuration file and remove the after=mpd constraint.

ThomasRausU, 2022-01-05:
ok, thank you - fully agree now.