MiczFlor/RPi-Jukebox-RFID #1708: jukebox-daemon.service as non root service

ThomasRausU, 2022-01-05:
At least in my setup (yocto) it's not possible to access user-mode PulseAudio (mixer controls) by jukebox-daemon when running as root.
I wonder if you discussed about running jukebox-daemon as non root in user-mode. Of course hardware (spi, gpio) access must be granted via user groups (spi, gpio). [e.g. like this or raspian style](https://forum.up-community.org/discussion/2141/solved-tutorial-gpio-i2c-spi-access-without-root-permissions)

ChisSoc, 2022-01-05:
Since Version 3.1.0, the Jukebox Service runs as user pi (in user mode) to connect to PulseAudio (same as MPD). Which version of future3 are you running? (Note: prior to  3.1.0 we did not use PulseAudio).

On my machine the service runs under user pi:
~~~
ps -elf | grep jukebox
0 S pi         685   603 15  80   0 - 94639 poll_s 17:09 ?        00:00:04 /usr/bin/python3 run_jukebox.py
~~~

ThomasRausU, 2022-01-05:
you named it. I installed this in the system mode directory and not user mode as you do with the current install script (since I can not use it with Yocto).
Now everything is running in user mode and my suggested changes are obsolete (beside nanoseconds).
thank you!

closed 2022-01-05 by ThomasRausU

ChisSoc, 2022-01-06:
Great! You might find [this part of the Docs helpful](https://rpi-jukebox-rfid.readthedocs.io/en/latest/userguide/system.html)