MiczFlor/RPi-Jukebox-RFID #1713: Small improvements on installation - future 3

s-martin, 2022-01-06:
I just installed from `future3/development` on a RPi 3B.

A few thoughts:

- Use some line breaks for the configuration questions so you know when the new question starts
- Provide an option to set the hostname as well
- make it more clear or ask for the static IP directly (I used the static IP, but kept the suggested IP)
- some kind of progress bar; "Updating Raspberry Pi OS" stays there for quite some time
- Is there an option to show the outputs of the installation like -verbose?
- a lot of npm packages seem to be deprecated
- 27 npm vulnerabilities, 16 moderate, 9 high, 2 critical mentioned
- I got this error in the install log, relevant?
```
Created symlink /etc/systemd/system/multi-user.target.wants/mpd.service -> /lib/systemd/system/mpd.service.
Created symlink /etc/systemd/system/sockets.target.wants/mpd.socket -> /lib/systemd/system/mpd.socket.
Job failed. See "journalctl -xe" for details
```

pabera, 2022-01-07:
Thanks Simon.. some thoughts inline

> Use some line breaks for the configuration questions so you know when the new question starts
> some kind of progress bar; "Updating Raspberry Pi OS" stays there for quite some time

Yes, this should be done. I did start a revamp of the installation process, but it takes time :-)

> Provide an option to set the hostname as well

I was researching this already, and it should be part of the installation process!

> make it more clear or ask for the static IP directly (I used the static IP, but kept the suggested IP)

Same here!

> Is there an option to show the outputs of the installation like -verbose?

Yes. Everything is logged onto the `INSTALL-XXX.log` file, you must have found it already. You can watch this file in a separate SSH session with `tail -f INSTALL-XXX.log`

> 27 npm vulnerabilities, 16 moderate, 9 high, 2 critical mentioned

The webapp depends on a lot of downstream dependencies we don't control. I have run an audit and provided a fix (#1714), so we are now at `8 moderate severity vulnerabilities`, which is a lot better.

> I got this error in the install log, relevant?

@ChisSoc, @ThomasRausU 
I think there was some discussion about this going on here -> #1703

ChisSoc, 2022-01-11:
> I think there was some discussion about this going on here -> #1703

Yes, I am on it.

s-martin, 2023-12-10:
was fixed

closed 2023-12-10 by s-martin