MiczFlor/RPi-Jukebox-RFID #1717: 🚀 | FEATURE SUMMARY

diiwill, 2022-01-07:
I currently have a working phoniebox on my home network and have several rfid cards with commands and songs linked.  I will soon be moving back on campus and will be using a wifi network that does not allow device-to-device communication. 

The raspberry pi setting a  static IP address and the web GUI will not be supported on this network. I only plan to use the cards I have already linked to songs and do not need to use the web GUI or connect new cards. 

Is there a way to bypass the pi setting its IP address so that I can use it headless on a public network? Will there be any other difficulties with this that I am not realizing? Thank you.

pabera, 2022-01-23:
The Phoniebox does not need network access to function, unless you require content from the web.
Does your box need access to the wifi, e.g. because you are using Spotify?

> Is there a way to bypass the pi setting its IP address so that I can use it headless 

If your public wifi allows to connect, then you need to remove the static IP address config and allow it to receive a dynamic IP via DCHP.

diiwill, 2022-01-23:
Yes, I use Spotify with my Phoniebox. 

> If your public wifi allows to connect, then you need to remove the static IP address config and allow it to receive a dynamic IP via DCHP.

Okay, thank you for the information. I will have a go at editing the dhcpcd.conf file.

closed 2022-01-23 by diiwill