MiczFlor/RPi-Jukebox-RFID #1721: setup of RC-522 not working anymore because of changed git protocol

golere, 2022-01-11:
When installing the RC522 with the setup script (/components/rfid-reader/setup_rc522.sh)
this error occurs:

`Installing Python requirements for RC522...
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
Command "git clone -q git://github.com/ondryaso/pi-rc522.git /tmp/pip-install-tkw78qk1/pi-rc522" failed with error code 128 in None
`
I'm trying to workaround this error by installing with pip
`pip3 install pi-rc522`

ChisSoc, 2022-01-11:
In the corresponding file the link needs to be changed from `git+git://` to `git+https://`

If you make the change and open a PR, I will merge that.

golere, 2022-01-11:
Hi ChisSoc,
thanks for your fast answer. I will check this out tomorrow, I never did a PR on github

ChisSoc, 2022-01-13:
A little misunderstanding #1722 was for Version 3. This Ticket for for Version 2.

lroeser, 2022-01-19:
install script for v2 doesn't use git protocols, it uses `pip install`:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/rfid-reader/RC522/setup_rc522.sh

lroeser, 2022-01-19:
`git+git:` still seems a legitimate protocol?:
`~$ sudo python3 -m pip install git+git://github.com/ondryaso/pi-rc522.git#egg=pi-rc522`
`Collecting pi-rc522 from git+git://github.com/ondryaso/pi-rc522.git#egg=pi-rc522`
`  Cloning git://github.com/ondryaso/pi-rc522.git to /tmp/pip-build-ahikq286/pi-rc522`

commit 7adacfc9 (2022-03-24): Fix git protocol for pip (#1791)

* Fix #1721

* Fix also spi installation
closed 2022-03-24 by s-martin