MiczFlor/RPi-Jukebox-RFID #1723: The number of my USB audio card change when I reboot and sometime stay the same

martinstjqc, 2022-01-12:
## Bug

The number of my USB audio card change when I reboot and sometime stay the same. I can see this change cause I don't have sound and if I do « aplay -l » in terminal I see the change... The number is always 1 or 2

EXAMPLE

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

### What I did and I expected this to happen

When I have this bug I change the number in asound.conf and alsa.conf at each time, but this is possible to fix the number for my USB audio device? Or to set 2 number in asound.conf and alsa.conf. I just want to not change the number in asound.conf and alsa.conf at each time.

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

On branch master
Your branch is up to date with 'origin/master'.

### Installscript

scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

3 B+

ChisSoc, 2022-01-20:
It's not a bug and not due to the Box' software. The ordering depends which card is found earlier during boot by Linux.

Edit your `boot/config.txt` to have this line (edit if it exits, add if not)
~~~
dtparam=audio=off
~~~
This disables on board audio

martinstjqc, 2022-01-21:
Thank a lot! I will try.

pabera, 2022-01-23:
I will close the ticket for now but feel free to reopen if you still struggle with your hardware.

closed 2022-01-23 by pabera