MiczFlor/RPi-Jukebox-RFID #1724: 🚀 | Start SH Script after positive CardID Verification

morpheus12, 2022-01-12:
## Feature Description
After positive verification of an card id (card was identified in database) a userscript should be executed for special usecases.
### What functionality would you like to see in your phoniebox?
I'm integrating phoniebox in the playroom. Here i have a Bose Soundtouch 20 Series II with original Bose API support which should be used as external audio amplifier (AUX Input). It works great but the Soundtouch switch to standby after 20min. Therefore i need to poweron again and switch to AUX Input.

Now i got a sh script to do the "Soundtouch stuff" and it works great (one-liner) :)

But how/where i should integrate the sh call?

### How do you envision the feature to work from a users perspective?

I will reduce the complexity for my children. All should work with card interaction. Therefore the soundsystem should be activated automatically. Thats why i want to use the soundtouch as active amp but the children should not interacting with the soundtouch directly (far away from their range due position very high in the room).

### Further information that might help

Currently i've added the sh call to rfid_trigger_play.sh directly after the echo command to store the current card id into the log. But here the script will executed by each reading interaction without consideration about the correct card-id. So a stop card will also activate the soundtouch as an correct playbook card id. Thats not cool.

THANKS for Feedback

pabera, 2022-01-12:
You could try to use a cron job. Once set up it would call your bash script regularly (e.g. every 15 min). 

- http://raspberry.tips/raspberrypi-einsteiger/cronjob-auf-dem-raspberry-pi-einrichten

morpheus12, 2022-01-14:
Well this would work, too. But i would like to activate it in usage only. Otherwise the soundtouch would be complete locked for other usage (webradio, airplay, etc). 

After 2 days in action it works ok for me. I've adopted the /scripts/playout_control.sh and /scripts/rfid_trigger_play.sh

**Playout_control.sh** in case of activation through webinterface (my first assumption was, that it would be enough here but it figured out that the rfid initiated play was not activating the script)

**rfid_trigger_play.sh** in case of card activation. It works but the script will activated each time when a card was read by the card-reader. It doesn't differentiate between registered cards and not registered cards. So i would like to place the script in case of a positive match of a card. But i don't get it.

pabera, 2022-01-23:
Great to hear it's working.
Since this seems to be resolved, I am closing the issue as well.

closed 2022-01-23 by pabera