MiczFlor/RPi-Jukebox-RFID #1727: Alias Missing

maedball, 2022-01-14:
Either the aliases inc and dec_volume are not done by now or the documentation is pointing to the wrong alias.

Invalid rpc command alias 'inc_volume' has been mapped to 'misc.empty_rpc_call()'

Groovylein, 2022-01-14:
@ChisSoc I think this is due to the move to Pulseaudio. Or did we forget to change it on the WebApp @pabera ?

Groovylein, 2022-01-14:
@maedball which version did you install?

maedball, 2022-01-14:
I just did a pull, so latest ;-).

all other aliases are working i am currently trying to change it away from alias to rpc, but since i am still working at the same time i havent gone far by now ;-p

    volume:
        Type: RotaryEncoder
        Pins: [6, 12]
        FunctionCW: {volume.ctrl.change_volume, kwargs: {step: 1}}
        FunctionCCW:  {alias: dec_volume,kwargs: {step: 1}}
        action_after_increments: 2

maedball, 2022-01-14:
It looks like i have it working, and it is just a documentation thing. 

    volume:
        Type: RotaryEncoder
        Pins: [6, 12]
        FunctionCW: {alias: change_volume, kwargs: {step: 5}}
        FunctionCCW:  {alias: change_volume, kwargs: {step: -5}}
        action_after_increments: 2

Groovylein, 2022-01-14:
> It looks like i have it working, and it is just a documentation thing. 

The template needs to be changed as well.

pabera, 2022-01-14:
> It looks like i have it working, and it is just a documentation thing.

@maedball: Great that you could figure it out yourself. Do you mind referring to the documentation that requires an update? :-)

pabera, 2022-01-14:
Just saw this #1730 .. 👍

maedball, 2022-01-14:
I am really bad in using Github so hope all is fine and it helps some ppl ;-)

By the way really nice and understandable structure. I wouldn't have figured it out without the well designed documents and structure! Now back to testing, next step gpio rfid reader...

closed 2022-01-14 by maedball

maedball, 2022-01-16:
> > It looks like i have it working, and it is just a documentation thing.
> 
> The template needs to be changed as well.

Sorry i just saw that comment, to what template are you referring?

Groovylein, 2022-01-16:
This one :)

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/resources/default-settings/gpio.example.yaml