MiczFlor/RPi-Jukebox-RFID #1740: Web Interface nicht mehr abrufbar von anderem Computer über die IP

Swissphonie, 2022-01-28:
Seit dem Wechsel auf unseren neuen WLan Router komme ich nicht mehr ins Web-Interface der Phonie Box

Das funktioniert:
Auf dem Raspberry habe ich Internet und kann über den lokalen Browser über die IP auf das Webinterface zugreifen wenn ich nen Monitor anschliesse.

Auf dem Fremdgerät (Mac osx) erscheint der Raspbi im Netzwerk, ich kann auch files hochladen.
Im Router erscheint der Raspi jedoch nicht (keine IP wird vergeben)
Über den Browser kann er keine Verbindung herstellen zum Interface (gleiche ip Adresse wie vom Pi aus läuft hier nichtl
Etwas verwirrend das Ganze. Bisher hat alles tip top funktioniert.
Hat jemand eine Idee woran das liegen könnte?
Wäre froh, wenn ich nicht immer die ganze Box aufschrauben und Monitor/Tastatur/Maus anhängen müsste um neue Titel aufzuspielen.
Besten Dank im Voraus für allfällig Hilfe.

Lippsson, 2022-02-01:
Hi Swissphonie!
Welche Browser hast du bisher ausprobiert? (Chrome, Edge, Firefox?)
Funktioniert es von einem anderen Rechner vielleicht?
Wenn du von deinem PC aus die Phoniebox pingen kannst würde ich es nochmal mit einer neuinstallation versuchen.
Probleme hatte ich auch schon mal Netzteilen/USB Kabel, es sollte keine Fehlermeldung wegen unterspannung zu sehen sein.

Swissphonie, 2022-02-03:
Hi,
danke für deine Tipps und sorry für die späte Antwort.
Browser hab ich Chrome und Safari auf dem Mac probiert.
Anderer Rechner teste ich noch, gute Idee!
Neuinstallation will ich nicht unbedingt, da ich als newbie lang gebraucht hab bis alles so schön lief wie jetzt mit den Buttons und so.
Netzteil zeigt’s nicht an, hängt aber schon viel Zeugs dran. Das werd ich auch noch testen.

Thx! schon mal!

Gruss

Lippsson, 2022-02-03:
Kann es sein das du 2 vershiedene Wlan´s hast (zb. Gäste und Eigenes)?
ûberprüfe auch mal ob der PI eine IP bekommt mit dem befehl "ifconfig" 
(https://www.elektronik-kompendium.de/sites/raspberry-pi/1906271.htm)

Swissphonie, 2022-02-06:
Hallo, ich kriege eine ip zugewiesen. Der Raspberry scheint mit dem W-Lan verbunden. Komischerweise kann ich zwar eine Google Suche machen, jedoch kann ich die Seiten wie z.B. gmx Mail oder github nicht öffnen da meint er diese seien unreachable.
Hier deshalb ein screenshot der ifconfig Ausgabe.

Ich habe bei meinem Modem/Router von Sagemcom in der Tat 2 Wlans gibt. Eines mit 2.4 G und eines mit 5 G. Diese sind so verhängt, dass der Router wohl das geeignetste für den aktuellen Zugriff aussucht? Evtl. liegt hier das Problem?

Anbei der Screenshot..

Herzlichen Dank fürs Anschauen, bin sehr froh um Tipps und Hilfe!

Gruss

Lippsson, 2022-02-08:
Hmm. ich seh keinen Screenshot. 
Da du anscheinend die Google suche erreichst, bist du mit einem Router verbunden. Das du ein paar Seiten nicht erreichen kannst, kann daran liegen das der Empfang schlecht ist und der Browser nach einer bestimmten Zeit abbricht. Das Wlan modul der Pi ist da auch nicht das beste und ich hab auch schon bei einigen eine USB-Wlan antenne gesehen. 2.4G und 5G spielt keine rolle, wichtig ist das der selbe IP-bereich a ist.
Wenn der Rapberry z.B. 192.168.1.50 hat, der PC 192.168.1.10 und Der Router als gateway 192.168.1.1 dann müsste alles funktionieren. Wenn der Raspbery allerdings 165.165.20.20 dann würde ich nochmal nachsehen wie man den Raspberry ins selbe Netzwerk bringen kann.

Swissphonie, 2022-02-08:
[image]

Lippsson, 2022-02-08:
Welche ip hat dein PC? Ich verwende kein IP6 im Raspberry, vielleicht kannst du mal im raspi-config diese testweise deaktivieren.

closed 2022-02-11 by s-martin