MiczFlor/RPi-Jukebox-RFID #1741: Default GPIO Button configuration conflicts with reserved GPIO pins

hgschwibbe, 2022-01-29:
## Bug

### What I did

I installed the Jukebox on an Raspberry Pi 3 using the Waveshare PN532 NFC HAT over I2C with a Shutdown button.

### What happened

After wiring the Shutdown button using the default PINS the NFC HAT was going out of work.

### I expected this to happen

I expected exactly this because the default Shutdown button configuration uses the I2C data wire at GPIO3 (SCL PC).

### Further information that might help
The default _gpio_settings.ini_ should not use GPIO pins which are reserved for the I2C, UART, or SPI protocol, see 
[Pin numbering on the RPi](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/img/GPIO-pin-numbering.png).

The green GPIO PINs are free to use, others are reserved for something, they should not be used.

And here is my working example that I am currently using:

`
[PrevNextControl]
enabled : True
type : TwoButtonControl
pin1 : 5
pin2 : 13
functioncall1 : functionCallPlayerPrev
functioncall2 : functionCallPlayerNext
functioncalltwobuttons : None
hold_time : 0.3
pull_up_down : pull_up
hold_mode : None

[PlayPause]
enabled : True
type : Button
pin : 21
hold_time : 0.3
functioncall : functionCallPlayerPause
pull_up_down : pull_up

[Shutdown]
enabled : True
type : Button
pin : 27
hold_time : 2
functioncall : functionCallShutdown
pull_up_down : pull_up
`

## Hardware

### RaspberryPi version
`3 B+`

### RFID Reader

Waveshare PN532 NFC HAT

T0bi79, 2022-12-29:
Relocating the default pin for the Shutdown button away from GPIO3 is a bad idea, since GPIO3 is the only pin which also allows to power up the pi again from a standby state by later pressing the same button.

s-martin, 2022-12-29:
@T0bi79 thanks for pointing that out.

From my point of view there is no perfect default configuration, so usually users need to adapt the GPIO pin settings to their individual setup.

Therefor I would close this issue for now as wontfix.

please feel free to open a new issue, if there’s a new problem.

closed 2022-12-29 by s-martin