MiczFlor/RPi-Jukebox-RFID #1742: 🚀 | Minimum volume setting

EarMaster, 2022-01-29:
## Feature Description

### What functionality would you like to see in your phoniebox?

The settings already have an option to control the upper limit of the volume, I propose a similar setting for the lower volume limit.

### How do you envision the feature to work from a users perspective?

My children - especially the younger one - sometimes just press the buttons for the fun of it. If they press _volume down_ too often and the volume reaches zero they sometimes think the box is "broken" when they later try to play something on it. A lower limit for the volume would prevent them from muting the box with the volume controls.

ChisSoc, 2022-02-07:
Was this feature request for Version 2 or Version 3?

EarMaster, 2022-02-07:
I am currently using Version 2, but I think it would be a good addition for both.