MiczFlor/RPi-Jukebox-RFID #1744: GPIO Default Behaviour after Reboot?

Karli2727, 2022-01-31:
I have a very trivial issue, I built a Box with GPIO buttons and it works like a charm. My child however is still very young and when we are playing something for her she randomly pushes buttons which can be quite annoying :-) So I disabled the gpio buttons on the web interface - which works fine. however after reboot the gpio settings return to being active, it does not remember the previous state. 

I also tried the setting the default option in the gpio_settings.ini to false but it did not do anyhting. I also tried to put every single buttons in the gpio_settings.ini to false but it had no effect upon reboot. 

Am I doing something terribly wrong here or is this a bug? 

All the best
Karl

Karli2727, 2022-02-09:
I'm sorry if I have done something wrong with submitting this issue, but is there a reason why I haven't gotten any reaction on it for the past 9 days? I saw that other issues were answered or at least labelled a couple of hours after they've been submitted. Did I break some code of conduct here?

pabera, 2022-02-12:
Hey Karl!

> I'm sorry if I have done something wrong with submitting this issue, but is there a reason why I haven't gotten any reaction on it for the past 9 days?

No one tries to avoid your post but usually people only answer when they really have something to say. And maybe that wasn't the case for your request.

The problem you describe is probably correct, and I don't have a solution for you now. This is mainly due to the fact that you are using 2.x but the current development focus is on 3.x.

I actually like the idea of switching of GPIO buttons on demand and of course, it should remain active after reboot as well. Why don't you tag this issue as `future3` and a feature request so it can be considered.

Unless you are able to fix it yourself, there might not be much support at the moment for such a feature in 2.x.

Karli2727, 2022-02-15:
Thank you very much for your reply. I understand the development focuses on 3.x and it would be great if it was considered there.  

I wouldn't classify this as a feature though because it already exists in the current GUI with the implemented Toggle Button, just that the state is not saved upon reboot which it should be because it is a general setting of the box.

AlvinSchiller, 2024-02-27:
Though this is a bit old, here would be a local solution to save the state even after reboot:

Open the file `htdocs/inc.header.php` and go to line 331 (starting with `gpiostatus`).
Add ` && /usr/bin/sudo /bin/systemctl enable phoniebox-gpio-control.service` to the `turnon` definition.
Add ` && /usr/bin/sudo /bin/systemctl disable phoniebox-gpio-control.service` to the `turnoff` definition.

The result should look someting like this:
```
    'gpiostatus' => array(
        'turnon' => "/usr/bin/sudo /bin/systemctl start phoniebox-gpio-control.service && /usr/bin/sudo /bin/systemctl enable phoniebox-gpio-control.service",                    // start the gpio control service
        'turnoff' => "/usr/bin/sudo /bin/systemctl stop phoniebox-gpio-control.service && /usr/bin/sudo /bin/systemctl disable phoniebox-gpio-control.service"                     // stop the gpio control service
    ),
```

With the `enable` / `disable` commands the state of the service is preserved on reboot.

AlvinSchiller, 2024-02-27:
Another idea for the described case, would be to set the `hold_mode` of the Buttons to `Postpone`. This delays the execution with `hold_time`. So you have to hold the buttons a while and not just single push them.