MiczFlor/RPi-Jukebox-RFID #1754: 🐛Trying to get RC522 to work Multi

switzel, 2022-02-07:
Hi,

I am trying to get the RC522 reader to work together with a USB reader. Accordingly I am using the `Reader.py.experimental.Multi` and `RegisterDevice.py.Multi` scripts. Each of the readers works fine on its own (using `Reader.experimental`) and the USB-Reader also works in the `.experimental.Multi` setup. But the RC522 doesn't.

The first minor problem is that `RegisterDevice.py.Multi` doesn't work but that is some weird python dependency problem that I think has nothing to do with this. Anyway, I think I figured out what the correct deviceName.txt should look like (for reasons not clear to me the `.Multi`s use a different format from the non-`.Multi`s) namely:

        MFRC522;
        Sycreader USB Reader;usb-3f980000.usb-1.1.2/input0

But then for some reason the RC522 is ignored. Has someone gotten this to work and has a suggestion?

Best,
Stefan

switzel, 2022-02-08:
Quick update (referring to Reader.py.experimental.Multi throughout): I just checked that everything high-level works as expected: the Reader registers the USB reader as well as the RC522 and polls each of their read functions.

I also checked that replacing Reader by Mfrc522Reader in `daemon_rfid_reader.py` works without any further changes (except removing the mysterious extra `reader.` in [https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/b3e91ae567cd3d089708f5b9d1a5f634bb41e8db/scripts/daemon_rfid_reader.py#L95]) what is that about?!). So is it possible that there is some synchronous stuff happening in Mfrc522Reader.readCard() that doesn't go well with multiprocessing?

switzel, 2022-02-08:
I think one might have to rewrite Mfrc522Reader.readCard() to not use RFID.wait_for_tag(). Then the reader could be stopped.

vitospinelli, 2022-08-16:
I get this error when I select my RC522 reader during the installation

phoniebox git clone -q git://github.com/ondtryaso/pi-rc522.git /tmp/pip-install-5yjpx8ai/pi-rc522 failed with error code 128 in None

vitospinelli, 2022-08-16:
Ok, manually installing RC522 following exactly these instructions solved the issue for me:

https://github.com/pimylifeup/MFRC522-python

s-martin, 2022-08-20:
Fixed by #1791, but not merged to master yet.

closed 2022-08-20 by s-martin