MiczFlor/RPi-Jukebox-RFID #1762: 🚀 | YouTube Music Ytmusic support

Tecnologic, 2022-02-12:
Hi, 

Play ytmusic songs/playlists/albums like Spotify.

For V2 I just installed the ytmusic extension to have get iris browsing my ytmusic account. When I then added the link from iris to an RFID card it worked. 

My question is where to start adding the feature? On which branch/fork can I start. How is Spotify integration planned for V3?

Thanks.

pabera, 2022-02-12:
> My question is where to start adding the feature? On which branch/fork can I start. How is Spotify integration planned for V3?

if you like to officially contribute to 2.x, you can send a pull request for the `develop` branch.
The primary backend to play MP3 is `mpd`. To allow seamless switching between local file playback and other backends, we are currently extending the `future3` player to be able to work with various music backends. [We currently work on Spotify](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/spotify) using the `librespot-java` plugin. 

This would allow us to integrate other backends as well, like Youtube Music. We could refer to mopidy's extension for [YT Music](https://github.com/OzymandiasTheGreat/mopidy-ytmusic).

Once we have released this multi music backend support, we can start adding more. If you like to contribute to the current Spotify branch, feel free to join.

MiczFlor, 2022-02-12:
I agree: if it works for V2.x, add it as a pull request to V2.x
future3 is the future, but still not the present :)

Tecnologic, 2022-03-18:
in the meantime i did dig a bit deeper. I came to the conclusion that using mopidy extension ytmusic with future3 is not a good solution because Spotify implementation uses librespot-java. But the ytmusicapi lib for python seems a valid option. 

The library gives back the stream url for a requested file. How can i forward that url to the jukebox player? 

Or asked differently where to start?

closed 2022-04-23 by Tecnologic