MiczFlor/RPi-Jukebox-RFID #1771: 🚀 | [future3] Documentation of Plugin System

Mischback, 2022-02-22:
## Feature Description

### What functionality would you like to see in your phoniebox?

This might be more of an documentation issue, nonetheless, I provide this as a feature request.

I'd like to provide some custom functionality for my Phoniebox. The *future3* plugin system is great, but it lacks documentation.

- Where to put custom code?!
- How to get it picked up by the actual Jukebox?
- (Best Practices for providing additional functionality)

### How do you envision the feature to work from a users perspective?

The default installation places the actual source in ``/home/pi/RPi-Jukebox-RFID``. That directory is under (Git) version control, which might not be totally desirable, but is fine for my purpose.

I'd like to place my own custom additions in a dedicated directory (let's say ``/home/pi/custom``), which might be a Git repository by itsself. Then I'd like to include my own plugins in the jukebox's configuration file.

### Further information that might help

I got my custom code picked up by placing it in ``~/RPi-Jukebox-RFID/src/jukebox/components/my_custom_extension`` and was able to interact with jukebox's provided code, i.e. using ``plugs``, the ``multitimer`` and the ``publishing`` interface. I can reference my code by adding a line in ``jukebox.yaml`` (``modules->named``) and can then use my custom code in the GPIO config.

However, mixing my own code with the official repository seems counter-intuitive.

I found #792, but that issue seems pretty much inactive.

s-martin, 2023-12-12:
The docstring in plugs.py contains info how to write plugins: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/src/jukebox/jukebox/plugs.py

s-martin, 2024-01-16:
Implemented in #2181

closed 2024-01-16 by s-martin