MiczFlor/RPi-Jukebox-RFID #1783: 🐛 | BUG SUMMARY - DietPi Bullseye - no RFID Reader

Surfer2010, 2022-03-05:
## Bug

### What I did

Software:
Started with DietPi - Bullseye to get way better boot times

Added User Pi + HomeDirectory
Added User DietPi and pi to the following groups: audio, gpio, sudo, www-data
(group plugdev neccessary as well?)

Added 2 packaged:
 - sudo apt-get install lighttpd-mod-deflate -y
 - sudo apt-get install python3-evdev -y

Installed Phoniebox with the following install script:

### What happened

1.) Webinterface Buttons on Settingspage "GPIO activate / RFID activate" do not react nor change anything, status shows disabled
2.) RFID Reader is not found nor can it be installed
[image]

### Further information that might help

RFID Reader Service Status:
[image]
GPIO Service Status:
[image]

## Software

### Base image and version

DietPi - Bullseye
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian

### Branch / Release

Version2.3 - 24258901 - master
EditionClassic edition (barebones)
cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
On branch master
Your branch is up to date with 'origin/master'.

### Installscript

Cd; wget https://github.com/MiczFlor/RPi-Jukebox-RFID/files/7534689/bullseye-install-default.zip && unzip bullseye-install-default.zip && ./bullseye-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi3 B

### RFID Reader

neuftech usb rfid reader

### Soundcard

Hifiberry 

### Other notable hardware

GPIO Extension Board + + onoffShim

s-martin, 2022-05-29:
Unfortunately Phoniebox relies heavily on user `pi`, see e.g. #1785.

so I assume this may be a problem you are facing.

s-martin, 2022-12-19:
Version 2.4 was released yesterday, which also includes #1543 and #1849.

@Surfer2010 you might want to check, if this fixes your issue.

AlvinSchiller, 2023-03-16:
Is the problem still present?
If yes could you post the installation log?

As in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1990 described, could some missing packages cause the installation of dependecies to fail, which then affects the service.

s-martin, 2023-12-03:
#2132 will fix user pi for future3.

For 2.x we won't fix this.

closed 2023-12-05 by pabera