MiczFlor/RPi-Jukebox-RFID #1785: Installation as different user (not user pi)

bigi-io, 2022-03-14:
## Feature Description
Right now, the installation routine does not work, if not run as the user pi

### What functionality would you like to see in your phoniebox?
it would be cool, if the installation would work without creating the user pi

### How do you envision the feature to work from a users perspective?
same installation procedure if logged in as other sudo-user

### Further information that might help
if not possible, it would be nice, if the installation guide is pointing to the necessity of creating user pi:
	sudo useradd pi
	sudo usermod -aG sudo pi

s-martin, 2022-04-08:
Might be necessary anyways

https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

s-martin, 2022-05-29:
@pabera is this adressed in future3?

Kreiszahler, 2022-09-13:
I would highly appriciate a version running with the latest RaspiOS.

s-martin, 2022-12-19:
Version 2.4 was released yesterday, which also includes PR #1543.

@bigi-io, you might want to check 2.4, if this fixes your issue.

mikea, 2023-01-21:
Still has user `pi`  backed in:

e.g.:
```
$ cat /etc/systemd/system/phoniebox-gpio-control.service
[Unit]
Description=Phoniebox GPIO Control Service
After=network.target iptables.service firewalld.service

[Service]
Type=simple
User=pi
Group=pi
```

mikea, 2023-01-21:
It is worth noting that fresh version of raspbian lite asks for a user-name as part of first boot. So it will be more and more likely not to have `pi` user.

AlvinSchiller, 2023-10-07:
@s-martin @pabera as this isn't a trivial change, should there be a check on installation if the current user is "pi" and quit otherwise? This would at least prevent upcoming issues of this kind until a proper solution is implemented.

s-martin, 2023-10-07:
I agree, a check on the name could at least give the user a hint what's going on.

In addition we should mention it in our docs that Phoniebox needs 'pi'.

AlvinSchiller, 2023-10-07:
> In addition we should mention it in our docs that Phoniebox needs 'pi'.
The current docs already mention that the User "pi" should / needs to be used.

s-martin, 2023-12-03:
#2132 will fix this for future3.

For 2.x we won't fix this.

closed 2023-12-05 by pabera