MiczFlor/RPi-Jukebox-RFID #1788: Register Device issue

GeneralMidi, 2022-03-17:
I'm trying to get a USB rfid reader registered using a rpi 1.

When running 

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py

I get

```
Choose the reader from list
0 Sycreader USB Reader
Device Number: 0
Exception ignored in: <function InputDevice.__del__ at 0xb6116270>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 160, in __del__
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 305, in close
  File "/usr/local/lib/python3.7/dist-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.7/asyncio/events.py", line 640, in get_event_loop
  File "/usr/lib/python3.7/asyncio/events.py", line 660, in new_event_loop
  File "/usr/lib/python3.7/asyncio/unix_events.py", line 51, in __init__
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 49, in __init__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 373, in __init__
  File "/usr/lib/python3.7/asyncio/coroutines.py", line 28, in _is_debug_mode
AttributeError: 'NoneType' object has no attribute 'dev_mode'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb614f8a0>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
```

dkiefner, 2022-03-18:
Hello, I am having the same issue. I am using:
- python version `3.7.3`
- Raspberry Pi OS: `Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`

It looks like an issue in the `evdev` module, but I am just guessing here 🤷.

SchulteMarkus, 2022-03-22:
Just updated Phonieboxand got the same issue.

petschni, 2022-03-22:
I had the same issue with a fresh install yesterday, but the following solved it for me: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1618#issuecomment-999682636

SchulteMarkus, 2022-03-23:
> I had the same issue with a fresh install yesterday, but the following solved it for me: [#1618 (comment)](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1618#issuecomment-999682636)

Helped!

closed 2022-03-24 by s-martin