MiczFlor/RPi-Jukebox-RFID #1799: Problem with RFID Reader🐛 | BUG SUMMARY

CoachPaulJohnson, 2022-04-03:
## Bug

### What I did
I reset my previously running Phoniebox (~Fall 2020) and installed the buster legacy OS. I ran the installer script. My RFID Scanner makes noises and conrol LED is on. Since the RFID scanner was previously working I expected it to work again.

### What happened

I was unable to add new cards with the web App. Further digging showed that I was unable to execute RegisterDevice.py to manually set up the RFID scanner. 
[image]

On another fresh install with more attention to detail this still prevailed.
[image]

## Software

### Installscript
scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version
4B

### RFID Reader

M301 Mifare card reader.

Thanks already!

ginandtronic, 2022-04-03:
HI Paul i had this very same problem with the Waveform PN532 tjis worked for me

`sudo apt-get install python3-evdev -y`

CoachPaulJohnson, 2022-04-03:
Hi and thanks alot!

This worked for me as well.

closed 2022-04-03 by CoachPaulJohnson