MiczFlor/RPi-Jukebox-RFID #1801: Spotify can not play some lists, search function does not work for spotify anymore

SirY4, 2022-04-03:
### Discussed in https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1786

<div type='discussions-op-text'>

<sup>Originally posted by **SirY4** March 15, 2022</sup>
Hi Phonieboxers,

on March 8 Spotify seen to have some major issues (see Twitter and other sources).  About that time I started to have some issues with my modipy: 
1.Some spotify playlists and links stopped working (webinterface and RFID triggered)
2. when I search things in the iris Modiy webinterface, I no longer get Spotify listings but only local options. 
I am not sure if the two are connected.

I get the message : (syntax error and could not load search result) "unexpected end of JSON input" within the IRIS web interface.

I upgraded with the one-line-installer but it did not change anything.

Do you have any ideas what to do?

Thank you!</div>

s-martin, 2022-04-04:
What’s the output of `journalctl -u mopidy`?

SirY4, 2022-04-04:
I solved the problem just now.
I previously updated the spotify client ID + secret manually in the config files. When rerunning the one-line-install-script I chose to keep the old settings and it used the ID and secret of the previous settings. 

I was not aware that the "keep current settings"-option in the install script would use its own configuration file instead of the current settings. 

Thank you your care for the project.

SirY4, 2022-04-04:
concerning the RFID-Problem: The recent patch  #1791 was very helpful for figuring out how to change the _requirements.txt_ files after the reinstall. I am looking forward to see the fix in the next release.  :-)

closed 2022-04-09 by SirY4