MiczFlor/RPi-Jukebox-RFID #1803: Not playing mp3s but Podcasts will play????🐛 | BUG SUMMARY

ginandtronic, 2022-04-05:
## Bug
Local files wont play but the podcasts in the example content will - I'm making this for my daughters birtday in 1 week and have spent hours getting it to function i finally heard joyous audio last night whilst shuffle playing the ZZZ example content and noticed the only files that will play are the podcast feeds....so strange??
### What I did
Intalled one line install buster as per guide, installed Waveform PN233 hat RFID reader, installed evdev & dependencies etc as per guide, RFID reader is working albeit after reinstalling it after boot. MPD fails to run on first boot, but works after sudo service mpd restart

### What happened
Only podcasts are played and not mp3s

### I expected this to happen
Upload and play mp3s

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version
3+B

### RFID Reader
Waveform PN232 Hat

### Soundcard
Headphone jack

### Other notable hardware

s-martin, 2022-04-05:
Are you using Spotify or classic edition?

ginandtronic, 2022-04-05:
Hi Martin i believe i installed both...

ginandtronic, 2022-04-06:
Its working great now with classic edition....

s-martin, 2022-05-29:
Unfortunately Spotify changed their API, so Spotify support is currently broken, see #1815

s-martin, 2022-05-29:
@ginandtronic If this works now, you could close this issue.

s-martin, 2022-12-18:
Closing this for now. Feel free to reopen, if the issue still exists.

closed 2022-12-18 by s-martin