MiczFlor/RPi-Jukebox-RFID #1806: Unable to change volume in future3 for a HifiBerry Mini amp (only max volume or mute)

marc-up, 2022-04-08:
Hi folks,

it's currently not possible to change the volume for the Hifi Berry Mini AMP using Buttons connected to the GPIO in future3.
The volume remains always on 'max' although the Web-GUI shows increasing or decreasing volume levels.

My setup is: Raspberry PI 2, Raspberry Pi OS Lite, Debian Version 10 (buster), Kernel 5.10 and jukebox in branch future3/main

I'm able to change the volume of the HiFiBerry Miniamp via
`paplay /usr/share/sounds/alsa/Front_Center.wav --volume=10000 `(=low volume) or
`paplay /usr/share/sounds/alsa/Front_Center.wav --volume=65536 `(=max volume).

I'm having a secondary audio output (USB-Soundcard) and it's possible to toggle the output via Web-GUI.
Furthermore I can and control the volume for the USB-Soundcard. But this is not possible for the HifiBerry Mini Amp. (Only 'mute' is working)

I wrote a post in the 'Discussions Forum'. After I figured out that there were other user with the same problem I think this should be marked as issue. 

I would love to contribute a fix for this issue, but currently I've no more idea where this issue results from.

ChisSoc, 2022-04-10:
This is curious thing is that I am using a HiFiBerry Mini Amp and a secondary sound output (but it is a bluetooth device), and I am having no trouble. 

First thing I notice, is that you are using buster - not bullseye. We have only tested with bullseye. Might be worth considering changing the OS release!

Are you able to change the volume from the command line, i.e. via alsamixer?

marc-up, 2022-05-20:
I think you can close the issue as it disappeared after a clean install on bullseye.

closed 2022-05-20 by marc-up