MiczFlor/RPi-Jukebox-RFID #1808: RC522 läuft nicht - "the unauthenticated git protocol on port 9418 is no longer supported."

and7891, 2022-04-16:
Hallo,
erstmal großes Lob für das tolle Projekt. Leider bekomme ich meinen RC522 nicht zum laufen. Ich bekomme einen Fehler wenn ich setup_rc522 ausführe. Bei der automatischen Installation bekomme ich den gleichen Fehler..
Würde mich freuen, wenn ihr mir weiterhelfen könnt. Hatte schon jemand das Problem? Gibt es einen workaround?

pi@raspberrypi:~ $ sudo python3 -m pip install --upgrade --force-reinstall -r /h                                                                                                             ome/pi/RPi-Jukebox-RFID/components/rfid-reader/PN532/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting py532lib
  Using cached py532lib-2018.6.14.dev1-py3-none-any.whl (34 kB)
Installing collected packages: py532lib
  Attempting uninstall: py532lib
    Found existing installation: py532lib 2018.6.14.dev1
    Uninstalling py532lib-2018.6.14.dev1:
      Successfully uninstalled py532lib-2018.6.14.dev1
Successfully installed py532lib-2018.6.14.dev1
pi@raspberrypi:~ $ sudo sh RPi-Jukebox-RFID/components/rfid-reader/RC522/setup_r                                                                                                             c522.sh
Please make sure that the RC522 reader is wired up correctly to the GPIO ports b                                                                                                             efore continuing...
Continue (y/n)? y
Installing Python requirements for RC522...
  ERROR: Command errored out with exit status 128:
   command: git clone -q git://github.com/ondryaso/pi-rc522.git /tmp/pip-install                                                                                                             -3y3mzmps/pi-rc522_d0ac81eb254d40b58e79ee7836b7f8d4
       cwd: None
  Complete output (3 lines):
  fatal: Fehler am anderen Ende:
    The unauthenticated git protocol on port 9418 is no longer supported.
  Please see https://github.blog/2021-09-01-improving-git-protocol-security-gith                                                                                                             ub/ for more information.
  ----------------------------------------
WARNING: Discarding git+git://github.com/ondryaso/pi-rc522.git#egg=pi-rc522. Com                                                                                                             mand errored out with exit status 128: git clone -q git://github.com/ondryaso/pi                                                                                                             -rc522.git /tmp/pip-install-3y3mzmps/pi-rc522_d0ac81eb254d40b58e79ee7836b7f8d4 C                                                                                                             heck the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pi-rc522 (unavail                                                                                                             able)
ERROR: No matching distribution found for pi-rc522 (unavailable)
Activating SPI...
Configure RFID reader in Phoniebox...
cp: der Aufruf von stat für '/home/pi/RPi-Jukebox-RFID/scripts/Reader.py.experim                                                                                                             ental' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
Restarting phoniebox-rfid-reader service...
Done.
pi@raspberrypi:~ $

danke.
and7891

s-martin, 2022-04-16:
Bitte die beiden Änderungen wie hier beschrieben machen:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791/files

s-martin, 2022-12-18:
V2.4.0 was released today - fixed.

closed 2022-12-18 by s-martin