MiczFlor/RPi-Jukebox-RFID #1809: gpio.yaml für V3.2.0 Beta

jens7885, 2022-04-21:
Moin! Ich habe die Box noch für eine frühe V2-Version gebaut mit einfachen Druckknöpfen und externen Widerständen.
Ich wollte nun mal V3.2.0 Beta ausprobieren und stehe in Sachen GPIO-Konfiguration ein bisschen da wie der Ochs vorm Berg.
Hätte jemand eine beispielhafte gpio.yaml für mich, an der ich mich orientieren kann? Wie gesagt, diese ganzen Features wie Drehknöpfe etc. sind alle nicht relevant und die Pin-Nummerierung kann ich anpassen. Danke!

ChisSoc, 2022-05-24:
Check out the [GPIO documentation](https://rpi-jukebox-rfid.readthedocs.io/en/latest/userguide/gpioz.html)

It will also point you to a sample file in your local installation `resources/default-settings/gpio.yaml`.

s-martin, 2022-12-27:
Closing for now.
Feel free to open a new issue, if the problem still occurs.

closed 2022-12-27 by s-martin