MiczFlor/RPi-Jukebox-RFID #1813: 🐛 | MOPIDY-Spotify is not installed

Reznap85, 2022-05-11:
### Bug
Hello all, I tried to install the phoniebox software several times with different Raspberry OS. (Currently Buster lite) Unfortunately i always have the same error (see below) with Mopidy-spotify and neither mp3 are playing nor Spotify can be connected. I have a spotify-family account (hope thats no problem).

### What I did
Fresh installation of BUSTER lite
configuration and tests of Sound and RFID working 
Several errors during installation:

### What happened
Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
**W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
E: The repository 'https://apt.mopidy.com buster InRelease' is not signed.
E: Unable to locate package libspotify-dev
Cloning into 'RPi-Jukebox-RFID'...
Installing dependencies for Spotify support...
Package: mopidy
Pin: version 3.*
Pin-Priority: 1001
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
E: The repository 'https://apt.mopidy.com buster InRelease' is not signed.
E: Unable to locate package mopidy-mpd
E: Unable to locate package mopidy-local
E: Unable to locate package mopidy-spotify
E: Unable to locate package libspotify12
E: Unable to locate package python3-spotify
Installing additional Python packages...
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.**
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
Command "git clone -q git://github.com/lthiery/SPI-Py.git /tmp/pip-install-j76eb_jt/spi-py" failed with error code 128 in None
Configuring Samba...
Configuring web server...
already enabled
Run "service lighttpd force-reload" to enable changes
already enabled
Run "service lighttpd force-reload" to enable changes
### Deleting older versions of service daemons. This might throw errors, ignore them
Failed to disable unit: Unit file idle-watchdog.service does not exist.
Failed to disable unit: Unit file rfid-reader.service does not exist.
Removed /etc/systemd/system/multi-user.target.wants/phoniebox-startup-sound.service.
Failed to disable unit: Unit file gpio-buttons.service does not exist.
Failed to disable unit: Unit file phoniebox-rotary-encoder.service does not exist.
Failed to disable unit: Unit file phoniebox-gpio-buttons.service does not exist.
rm: cannot remove '/etc/systemd/system/rfid-reader.service': No such file or directory
rm: cannot remove '/etc/systemd/system/startup-sound.service': No such file or directory
rm: cannot remove '/etc/systemd/system/gpio-buttons.service': No such file or directory
rm: cannot remove '/etc/systemd/system/idle-watchdog.service': No such file or directory
rm: cannot remove '/etc/systemd/system/phoniebox-rotary-encoder.service': No such file or directory
rm: cannot remove '/etc/systemd/system/phoniebox-gpio-buttons.service': No such file or directory
### Done with erasing old daemons. Stop ignoring errors!
Configuring Spotify support...
Synchronizing state of mpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable mpd
Failed to enable unit: Unit file mopidy.service does not exist.
Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
cp: cannot create regular file '/etc/mopidy/mopidy.conf': No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
Configuring MPD...
mpd error: Connection refused
Starting dhcpcd service...
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
Setting owner and permissions for directories...
amixer: Control default element write error: Operation not permitted

#
# INSTALLATION FINISHED

### I expected this to happen

Can please somebody help with a workaround or fix? thank you! I've searched a lot and cannot find any solution. Thank you!

### Further information that might help

 'journalctl -u mopidy' ( Spotify edition only)
-- Logs begin at Wed 2022-05-11 20:17:01 BST, end at Wed 2022-05-11 21:42:30 BST. --
-- No entries --

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

### Branch / Release

On branch master
Your branch is up to date with 'origin/master'.

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Plus Rev 1.3

### RFID Reader

16c0:27db HXGCoLtd Keyboard`

### Soundcard

Behringer UCA202

### Other notable hardware

s-martin, 2022-05-11:
This should fix the issue https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1792

But this is not yet in the latest release.

erickeller, 2022-11-19:
any hint when the next release will include this patch?
As the readme specifies that Spotify is not supported anymore, maybe one could simply remove these dependencies?

> IMPORTANT NOTICE REGARDING SPOTIFY INTEGRATION
> Spotify has [disabled access to libspotify on May 16, 2022](https://developer.spotify.com/community/news/2022/04/12/libspotify-sunset/). This means Phoniebox can not communicate with Spotify via libspotify (and mopidy-spotify) anymore. The problem is not our code but the cut off by Spotify.

s-martin, 2022-12-18:
This is a duplicate of #1815.

Please follow the thread and possible solutions there.

I close this issue now as a duplicate, but feel free to reopen it.

closed 2022-12-18 by s-martin