MiczFlor/RPi-Jukebox-RFID #1814: Websocket not working because of failed zmq install

fgma, 2022-05-13:
### Describe your problem 

After isntalling webinterface does not work as expected. All dynamic elements show a loading animation and the websocket connection fails. Logs show some problem concerning zmq. Installation of zmq was not successfull in the first place.

```
pi@raspberrypi:~ $ journalctl --user -b -u jukebox-daemon
-- Journal begins at Mon 2022-04-04 14:06:38 CEST, ends at Fri 2022-05-13 20:04:06 CEST. --
May 13 19:45:42 raspberrypi systemd[591]: Started Jukebox Daemon.
May 13 19:45:44 raspberrypi python3[622]: 13.05.2022 19:45:44 -   41:daemon.py          - jb.daemon            - MainThread      - INFO     - Starting Jukebox Daemon (Version 3.2.0)
May 13 19:45:44 raspberrypi python3[622]: 13.05.2022 19:45:44 -   44:daemon.py          - jb.daemon            - MainThread      - INFO     - Git state: 8071023 [2022-02-10] Release 3.2.0  (HEAD -> future3/main,>
May 13 19:45:44 raspberrypi python3[622]: 13.05.2022 19:45:44 -  317:cfghandler.py      - jb.cfghandler        - MainThread      - INFO     - (jukebox) Loading yaml file '/home/pi/RPi-Jukebox-RFID/shared/setting>
May 13 19:45:44 raspberrypi python3[622]: 13.05.2022 19:45:44 -  230:cfghandler.py      - jb.cfghandler        - MainThread      - DEBUG    - (jukebox) Replacing current config data
May 13 19:45:44 raspberrypi python3[622]: 13.05.2022 19:45:44 -   53:daemon.py          - jb.daemon            - MainThread      - INFO     - Welcome to Jukebox
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -   54:daemon.py          - jb.daemon            - MainThread      - INFO     - Time of start: Fri May 13 19:45:44 2022
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  572:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.publishing' as 'publishing'
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  304:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting function 'components.publishing.republish' as 'publishing.r>
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  597:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Package load initializer: calling publishing.initialize()
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  615:plugs.py           - jb.plugin            - MainThread      - ERROR    - Ignoring failed package load for 'publishing'
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  616:plugs.py           - jb.plugin            - MainThread      - ERROR    - Reason: ZMQError: Protocol not supported
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  617:plugs.py           - jb.plugin            - MainThread      - ERROR    - Detailed reason:
May 13 19:45:45 raspberrypi python3[622]: Traceback (most recent call last):
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 612, in load_all_named
May 13 19:45:45 raspberrypi python3[622]:     load(package, load_as, prefix)
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 602, in load
May 13 19:45:45 raspberrypi python3[622]:     raise e
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 599, in load
May 13 19:45:45 raspberrypi python3[622]:     func()
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/publishing/__init__.py", line 36, in initialize
May 13 19:45:45 raspberrypi python3[622]:     _PUBLISH_SERVER_THREAD = pub.server.PublishServer(tcp_port=tcp_port, websocket_port=ws_port)
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/publishing/server.py", line 223, in __init__
May 13 19:45:45 raspberrypi python3[622]:     self.backend.bind(f'ws://*:{websocket_port}')
May 13 19:45:45 raspberrypi python3[622]:   File "/home/pi/.local/lib/python3.9/site-packages/zmq/sugar/socket.py", line 214, in bind
May 13 19:45:45 raspberrypi python3[622]:     super().bind(addr)
May 13 19:45:45 raspberrypi python3[622]:   File "zmq/backend/cython/socket.pyx", line 540, in zmq.backend.cython.socket.Socket.bind
May 13 19:45:45 raspberrypi python3[622]:   File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
May 13 19:45:45 raspberrypi python3[622]: zmq.error.ZMQError: Protocol not supported
May 13 19:45:45 raspberrypi python3[622]:
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  572:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.volume' as 'volume'
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  487:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.toggle_output
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  487:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.get_outputs
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  487:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.publish_volume
May 13 19:45:45 raspberrypi python3[622]: 13.05.2022 19:45:45 -  487:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.publish_outputs
```

```pi@raspberrypi:~ $ pip list
Package                       Version
----------------------------- ---------
alabaster                     0.7.12
attrs                         21.4.0
Babel                         2.10.1
certifi                       2020.6.20
chardet                       4.0.0
colorzero                     1.1
coverage                      5.5
deprecation                   2.1.0
distro                        1.5.0
dnspython                     2.0.0
docutils                      0.17.1
evdev                         1.5.0
eyed3                         0.9.6
filetype                      1.0.13
flake8                        4.0.1
gpiozero                      1.6.2
idna                          2.10
imagesize                     1.3.0
importlib-metadata            4.11.3
iniconfig                     1.1.1
Jinja2                        3.1.2
MarkupSafe                    2.1.1
mccabe                        0.6.1
mock                          4.0.3
mutagen                       1.45.1
packaging                     21.3
pip                           22.1
pluggy                        1.0.0
pulsectl                      22.3.2
py                            1.11.0
pyalsaaudio                   0.9.2
pycodestyle                   2.8.0
pyflakes                      2.4.0
Pygments                      2.12.0
pyparsing                     3.0.9
pytest                        7.1.2
python-apt                    2.2.1
python-mpd2                   3.0.5
pytz                          2022.1
pyzmq                         22.3.0
requests                      2.25.1
RPi.GPIO                      0.7.0
ruamel.yaml                   0.17.21
ruamel.yaml.clib              0.2.6
setuptools                    52.0.0
six                           1.16.0
snowballstemmer               2.2.0
Sphinx                        4.5.0
sphinx-rtd-theme              1.0.0
sphinxcontrib-applehelp       1.0.2
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        2.0.0
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.5
spidev                        3.5
ssh-import-id                 5.10
toml                          0.10.2
tomli                         2.0.1
tornado                       6.1
urllib3                       1.26.5
wheel                         0.34.2
zipp                          3.8.0
pi@raspberrypi:~ $```

#### What's your hardware set up? 

Pi 4. Hifiberry Miniamp, for testing no RFID reader

#### If possible, try to attach logs from ... (paths from RPi)

[INSTALL-1652463300.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8689954/INSTALL-1652463300.log)

KrystianNiepsuj, 2022-05-16:
I received the same error. 

1. I had just done a clean install of raspbian on my raspberry pi 3 - 
2. I immediately went to install phoniebox / future3 using the link provided here [https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html)
3. Tried to set up my RFID reader by running  `src/jukebox/run_register_rfid_reader.py`
4. Received an exception on install that pyzmq is not installed. I installed the latest by running `pip3 install pyzmq`
5.  I was then able to succesfully register my PN532 RFID reader
6.  When browsing the webapp everything appears to be loading and i received the identical message above

KrystianNiepsuj, 2022-05-17:
My mistake @Lippsson  - i had inadvertently left a typo in the comment. I did not have time yet to test on a fresh raspberry pi - so your mileage may vary a little bit with the updated pyzmq build.

s-martin, 2023-11-18:
Should be now fixed on `future3/develop`

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2108#issuecomment-1809167778

can you check this?

closing for now. Feel free to post in #2108, if the issue still occurs.

closed 2023-11-18 by s-martin