MiczFlor/RPi-Jukebox-RFID #1815: 🐛 | Spotify does not working anymore

hyperbit00, 2022-05-16:
## Bug

### What I did

My Working Phoniebox stops working since today.
Spotify integration does not work. only local playback works

I Changed nothing. Works weeks before like a charm.
Spotify Premium is Valid

### What happened

May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.2.0
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.config Loading config from command line options
May 16 19:46:47 phoniebox mopidy[2207]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, spotify, file, mpd, iris, m3u, stream, softwaremixer, http
May 16 19:46:47 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
May 16 19:46:48 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
May 16 19:46:48 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
May 16 19:46:48 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
May 16 19:46:48 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
May 16 19:46:49 phoniebox mopidy[2207]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.USER_NEEDS_PREMIUM: 15>
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as XXXXXXXXXX
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting Mopidy core
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.54.2
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [0.0.0.0]:6680
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [0.0.0.0]:6600
May 16 19:46:49 phoniebox mopidy[2207]: INFO     [MainThread] mopidy.commands Starting GLib mainloop

May 16 19:49:29 phoniebox mopidy[2207]: INFO     [MpdSession-278] mopidy_mpd.session New MPD connection from [127.0.0.1]:34620
May 16 19:49:29 phoniebox mopidy[2207]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:album:0XXX': Session must be logged in and online to load objects: <ConnectionState.LOGGED_O
May 16 19:49:29 phoniebox mopidy[2207]: INFO     [MpdSession-279] mopidy_mpd.session New MPD connection from [127.0.0.1]:34622

hyperbit00, 2022-05-16:
I think its a problem discribed here -> ? https://github.com/mopidy/mopidy-spotify

"As [promised last month](https://developer.spotify.com/community/news/2022/04/12/libspotify-sunset), Spotify disabled access to libspotify on 16/05/2022. As of today, seven years after libspotify was deprecated, there is still no official replacement library available."

s-martin, 2022-05-16:
Thanks for reporting this.

Looks like this is a real showstopper.

Vanierica, 2022-05-18:
Can confirm. We have 2 Phonieboxes for our children and both stopped playing spotify over night.

thomas-wolf, 2022-05-18:
Confirmed by 5 boxes and 5 sad children. There will be no quick way around it. We all hoped that day would never come, but it finally has come and there is not much one can do as long as spotify does not reconsider its decision. At least the non-spotify part is still working.

s-martin, 2022-05-18:
> Holy shite, yes please prioritized this, I think this breaks every other phoniebox out there (including mine :/)

Unfortunately I don’t think that there will be a solution to this, because Spotify disabled the support of libspotify.

s-martin, 2022-05-18:
@MiczFlor

DavidOrtmann, 2022-05-18:
Is it out of the question to use another API or SDK for Spotify on Phoniebox? It's really depressing to lose that unique selling point. I didn't see it coming, either.

s-martin, 2022-05-18:
> Is it out of the question to use another API or SDK for Spotify on Phoniebox? It's really depressing to lose that unique selling point. I didn't see it coming, either.

See https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1815#issuecomment-1130384442

sonicdee, 2022-05-20:
^that might work, see how to install https://qengineering.eu/install-gstreamer-1.18-on-raspberry-pi-4.html
( I hadn't found time to try ;(  )

s-martin, 2022-05-20:
> Same here spotify is not working anymore. Is there any alternative API or SDK to replace mopidy?
> 
> whats about this: https://lib.rs/crates/gst-plugin-spotify (found here [mopidy/mopidy-spotify#110 (comment)](https://github.com/mopidy/mopidy-spotify/issues/110#issuecomment-1129917534))

Current implementation relies on mopidy in the Spotify edition (also for non Spotify functions), so from my point of view we should wait for an update of mopidy-spotify. 

But if anyone comes up with a different solution and proposes a PR, we could also check if this is the way we want to go.

As we are currently working on a new and shiny version 3 we need to check the implications in https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/spotify as well.

@pabera @ChisSoc @Groovylein @MiczFlor

sonicdee, 2022-05-20:
> As we are currently working on a new and shiny version 3 we need to check the implications in https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/spotify as well.

^^I've seen there is already alottah python inside, this might be handsome, I'll try...

https://pypi.org/project/spotify/

Groovylein, 2022-05-20:
> As we are currently working on a new and shiny version 3 we need to check the implications in https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/spotify as well.

I've tested the development environment for Spotify yesterday and it works 🤞 
We are using https://github.com/librespot-org/librespot-java, which is using the Spotify Web-API

sonicdee, 2022-05-20:
> I've tested the development environment for Spotify yesterday and it works 🤞 We are using https://github.com/librespot-org/librespot-java, which is using the Spotify Web-API

yeah, would be so nice to have a kind of working version, so guys can swtich to future 3 ;)

usamer77, 2022-05-20:
Hey Guys, that would be so so so awesome if you are able to find an alternative solution! You would be a star for so many Kids and Dads! :-)

morpheus12, 2022-05-20:
I'm to stupid for active coding contribution. It would be awesome to get an alternative for my son. We both created a huge amount of cards with his favorite audiobooks. Therefore it would be great if you professionals find a way to solve the problem.

s-martin, 2022-05-20:
> I've tested the development environment for Spotify yesterday and it works 🤞 We are using https://github.com/librespot-org/librespot-java, which is using the Spotify Web-API

How mature is the future3/spotify branch? Can it be merged and released yet?

Groovylein, 2022-05-21:
> How mature is the future3/spotify branch? Can it be merged and released yet?

Unfortunately it isn't stable at all. The player management and the whole WebApps are not ready for Spotify yet.

lenlennart, 2022-05-21:
Can we get a merge of future3/dev and spotify? Even if not stable and no web-app, i guess you will get a lot of testers :-)

ChisSoc, 2022-05-21:
A few words about the future3/spotify branch: This is not just adding Spotify. This branch massively changes (and improves) the way the Jukebox interfaces with the backend players. Previously there has only been one: MPD, and now there need to be multiple. So even future3/develop and future3/spotify are not compatible at the moment and need to be merged, which is a bit of work. For the user, this is also a not quite transparent change: the way stuff (links etc) is stored must to change from the current release. Currently the spotify branch is also an experimental branch for all these underlying changes. And that is where it is: experimental - many changes, no documentation, little test coverage.

After a highly active coding phase during winter with much progress, there has been a bit of a lush with the beginning of spring. Guess why ;-) We need to pick it up from there. So, long words short: there is no quick solution. It is massively more work than just merging the current branch to develop.

Micharius, 2022-05-21:
Just came to the discussions to find out, why my daughter's Phoniebox stopped working. And now - oh my, what a catastrophe 😥, our Phoniebox is entirely based on Spotify, I made over 50 cards.  Hopefully, it will work again in Future, shame on Spotify to just cut the access.

maahn, 2022-05-21:
This was quite shocking news so I thought about alternatives out of desperation: I have an LMS SqueezeboxServer server on another Raspberry Pi (with working Spotify support), so I installed piCorePlayer on my Phoniebox to use it as an LMS Squeezebox Client, and wrote a script to use RFID tags for starting Spotify playlists. 

Of course, this is not supposed to be a replacement for Phoniebox, just a quick and dirty fix to get Spotify working again quickly.  I hope you find a better solution soon! Until then, if someone is desperate enough to try it out, the script and instructions are here https://github.com/maahn/piCorePhonie

sonicdee, 2022-05-22:
> wrote a script to use RFID tags for starting Spotify playlists.
nice!! 
I'm on a just python script, back to basics, I started yesterday, reading the usb rfid number works,
have less time but will be ready hopefully soon

lenlennart, 2022-05-24:
> A few words about the future3/spotify branch: This is not just adding Spotify. This branch massively changes (and improves) the way the Jukebox interfaces with the backend players. Previously there [...]

hmm here are some thoughts from my side: Instead of trying to create an Eierlegende Wollmilchsau, maybe be should take a step back and aim for a simple back-to-the-roots version: A codebase that allows playing of spotify based on RFID cards.

sonicdee, 2022-05-24:
oh deer ;)
it kind of works, but not really ;)
daughter is happy I converted her box to a just Spotiy Connect Client with raspotify
she's using her laptop to start an album and closes it after it's playing through the pi -> audio jack -> partybox
works flawless since 2 days ;) ... she's old enough to get in touch to the whole spotify world...

for her little brother -> rfid cards kind of work ... 
see the gist, feel free to do any better ;) I'm not a soo experienced python guy ...
https://gist.github.com/sonicdee/bf5f655669ef6900b72c54f0c7696d32
it's using raspotify also, and spotipy to control it, 
but
I'm running in an issue the raspotify device is not active on next day, so it can't be controlled anymore ;( 
^^ maybe it can be done all inside librespot better !?

I stuck on this device loosing problem - reboot seems to work
also haven't figured out how to use py asyncio, but using screen with crontab to start both - works also ;) .. 
you need to start gpio and Rfspoty side by side
also -> sudo is needed for usb rfid, can't find the right stuff to change udev rules...

s-martin, 2022-05-29:
https://www.heise.de/news/Abgeschaltet-libspotify-Aus-legt-selbstgebastelte-Musikboxen-fuer-Kinder-lahm-7124231.html

thisismana, 2022-05-30:
After getting an glimpse overview of why Spotify stopped working, I quit my Premium subscription right away (as this was the only reason to have a subscription in the first place). I cannot imagine that they will revive this API — ever. Guess this is expected to happen if you have the monopoly on audio-streaming?!

Has anyone got their box working with providers other than Spotify?

shinebar1001, 2022-05-30:
I'll stick with spotify and converted all my stuff to mp3. Slightly more effort but it works even offline.

oswaldo-be, 2022-05-30:
I switched to MuPiBox and bought touchscreens as adviced... but it only worked one time to play spotify. Now I have a problem.. I hope it will be solved

varac, 2022-05-30:
@oswaldo-be Could you please share more of your migration experiences with MuPiBox ? Given the time frame of a possible fix for Phoniebox I guess it's a viable alternative to many of us, as sad as it is.

s-martin, 2022-05-30:
> After getting an glimpse overview of why Spotify stopped working, I quit my Premium subscription right away (as this was the only reason to have a subscription in the first place). I cannot imagine [...]

In our new version 3 we use a different implementation of the Spotify API, but the branch future3/spotify is not ready for production yet, see https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1815#issuecomment-1132774819

danielsmint, 2022-05-31:
> I'll stick with spotify and converted all my stuff to mp3. Slightly more effort but it works even offline.

I did this in the first place, had some connectivity issues. But its realy sad, i hope there will be a alternative too. Maybe with offline playlists? (i hope)

sonicdee, 2022-05-31:
> I stuck on this device loosing problem - reboot seems to work ^^ I'm investigating these days, when it works I'll load it to github with better docs...
I've added "sudo systemctl restart raspotify" to cronjob and this seems to work... for a couple of days now ;) happy kids

s-martin, 2022-06-01:
Please follow the discussion about the future of Spotify integration in #1835

Groovylein, 2022-06-02:
As you already all know spotify discontinued libspotify library (see [README.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/README.md#phoniebox-the-rpi-jukebox-rfid)) and lobrespot-java added a [disclaimer](https://github.com/librespot-org/librespot-java#disclaimer).

For this very reason, we had to delete the branch future3/spotify, since we are now searching for new solution approaches to integrate spotify API into the future3 phoniebox.

Feel free to propose new ways of integrating supported spotify API into the phoniebox in the discussion forum #1835

passuff, 2022-06-02:
> i just did the same and frankly, this wasnt too much work - roughly 2min
> for each of the 20 cards.
> Me - and more importantly my daughter - is happy again.
> looking forward to see a new spotify integration, but the immediate pain
> is gone.

Which tool did you use to convert it?

rmk-ch, 2022-06-02:
Hi phonieboxers.

I would like to share two solutions/workarounds to satisfy our children until the ultimate Phoniebox release.

1. Download albums/playlists as MP3 and move them to the phoniebox SD-card. In my opinion the best way to achieve this is using https://github.com/SathyaBhat/spotify-dl/ and following the description there.
2. And the solution I prefer: Delete the premium account of spotify and create one with [TIDAL](https://tidal.com). The look and feel is almost the same as with spotify and works for all of my setups (Sonos integration, android cell phone incl. offline mode and locally on a Windows PC). Moreover, I was able to move my playlists and favorites from Spotify over to Tidal using [this service](https://tidal.com/import-playlist). On the Phoniebox you need to do the following steps:
  2.1. Install [mopidy-tidal](https://github.com/tehkillerbee/mopidy-tidal) according to the Readme. Follow the procedures to for [configuration](https://github.com/tehkillerbee/mopidy-tidal) and [Oauth authentification](https://github.com/tehkillerbee/mopidy-tidal) (see log files for link and authentificate). 
  2.2. In the Phoniebox webinterface go to Iris (by clicking "Spotify" in the top menu)
  2.3. You can search for whatever album/playlist you would like to link to an RFID card. Tidal streams should be coming up then! Copy that URL (something like tidal:album:163359476). (Note the link must come from the Iris webinterface, not from the Tidal app!)
  2.4. Link the card to that URL in the Phoniebox webinterface as you would have done previously with any spotify album/playlist. Create a new folder with a specific name and select "Spotify" as stream type (this is important!).
  2.5. Done. This should work then. Check the log files of mopidy if you experience issues (`journalctl -u mopidy`)

Hope this helps you as well!
Best,
rmk

lenlennart, 2022-06-02:
> Which tool did you use to convert it?

i used tunefab, 30 day trail with 3min limit per song.
So not really usable for audio books, but fine for songs.

varac, 2022-06-03:
>     * And the solution I prefer: Delete the premium account of spotify and create one with [TIDAL](https://tidal.com). The look and feel is almost the same as with spotify and works for all of my setups (Sonos integration, android cell phone incl. offline mode and locally on a Windows PC). Moreover, I was able to move my playlists and favorites from Spotify over to Tidal using [this service](https://tidal.com/import-playlist). On the Phoniebox you need to do the following steps:
>       
>       1. Install [mopidy-tidal](https://github.com/tehkillerbee/mopidy-tidal) according to the Readme. Follow the procedures to for [configuration](https://github.com/tehkillerbee/mopidy-tidal) and [Oauth authentification](https://github.com/tehkillerbee/mopidy-tidal) (see log files for link and authentificate).
>       2. In the Phoniebox webinterface go to Iris (by clicking "Spotify" in the top menu)
>       3. You can search for whatever album/playlist you would like to link to an RFID card. Tidal streams should be coming up then! Copy that URL (something like tidal:album:163359476). (Note the link must come from the Iris webinterface, not from the Tidal app!)
>       4. Link the card to that URL in the Phoniebox webinterface as you would have done previously with any spotify album/playlist. Create a new folder with a specific name and select "Spotify" as stream type (this is important!).
>       5. Done. This should work then. Check the log files of mopidy if you experience issues (`journalctl -u mopidy`)

@rmk-ch That sounds great! Could you please share what Phoniebox version you use ? I guess it's v2, because I don't see any mopidy in current future3 branch/v3.2.0 release, or am I wrong ?
Is there any chance to easily add mopidy to future3 ?

rmk-ch, 2022-06-04:
@varac I am running v2 with spotify enabled.
@passuff ouu i see you're right about this. This is not a spotify download. Sorry for the confusion. Thought it was mich better due zo lower wait times compared to many other tools. I haven't tested this thoroughly due to switching to tidal.

usernameAXZ, 2022-06-07:
> Which tool did you use to convert it? 

I can recommend [DownOnSpot](https://github.com/oSumAtrIX/DownOnSpot) for this purpose. It is based on librespot and requires a nightly Rust toolchain for building, but overall the process is fairly straightforward. They claim it will also work with a free Spotify account, I did not verify this, but it definitely worked well with my premium account. 
Actually now I like this solution more than the streaming version, it is more reliable and Spotify can no longer mess with my custom playlists (which they did a couple of times in the past). Not sure if its covered by their TOS but as long as you have a premium account, it could be argued this falls under fair use.

passuff, 2022-06-07:
Thx. Will have a deep look into it.

kingosticks, 2022-06-08:
Hi guys, I just wanted to say a couple of things. 

It sucks Spotify have turned libspotify off and they've no interest in supporting projects like this. I assume some of the comments on the Spotify dev forum are from (rightly) unhappy Phoniebox users and although I don't think Spotify will reverse this move, it's still good to demonstrate there is demand for a real libspotify replacement. So thanks to everyone who posted.

Secondly, I am working on a fixed version of Mopidy-Spotify but I don't have much free time so it's sporadic development. And it's a fair bit of work. But progress is being made and I hope to get it working again soon. I can't make any specific timescale promises. I would understand if Phoniebox was to move to another software solution, it's important to find the best fit for your project and get things working for users. But I would hope you avoid projects that circumvent Spotify's Premium subscription requirement as I think that'll just end in everyone losing access. Alternatives services like Tidal are an option but they have even worse developer support (it's non-existent). No good options out there, sadly. 

I also want to mention it was my colleague that randomly reminded me of Phoniebox when suggesting it as a cool project for me, now I'm a father. Hopefully this explains why dev is slow but at least I've got a vested interest now!

usamer77, 2022-08-08:
Hi Guys, just wanted to catch up on the last activities since June. What is the actual status? Is there a workaround in place for the libspotify workaround? 

Many thanks!

The-Judge, 2022-08-09:
283 cards and 2 Phonieboxes deep into this project ... It would be REALLY great if a working release, at least with RFID card reader + working Spotify support, would be released. Even as pre-pre-pre-alpha without any working Web-Interface or anything would be highly appreciated. RFID + Spotify are the only features we use anyways.

Any roadmap there?

s-martin, 2022-08-20:
Hi, discussion of Spotify feature in Phoniebox should be done in #1835.

As of today there’s no solution yet.

The previously used mopidy-spotify implementation seems to make some progress, but no solution yet:
https://github.com/mopidy/mopidy-spotify/issues/110

MrStrategy, 2022-08-24:
Hey all,

while this is not yet a perfect solution I got spotify working again on phoniebox.  
The solution is based on the thread referenced by @s-martin.
Major thanks goes to all the contributors in the [https://github.com/mopidy/mopidy-spotify/issues/110 thread](https://github.com/mopidy/mopidy-spotify/issues/110).

**be advised: The solution requires some minor code changes and compilation of one software module** , so you should have some basic Linux knowledge to apply the workaround.  
I've applied the solution on a Raspberry Pi 4 and slight adjustments might be necessary on older PIs due to the different ARM-architecture.

What did I do to get the solution working:  
1. Download and install a precompiled version of the so-called spotify url handler for gstreamer. This is a specific version of the handler as it was adjusted to be mopidy compliant.  
```
wget https://github.com/mopidy/mopidy-spotify/files/9176522/gst-plugin-spotify_0.9.0_armhf.deb.zip
unzip https://github.com/mopidy/mopidy-spotify/files/9176522/gst-plugin-spotify_0.9.0_armhf.deb.zip  
sudo dpkg -i gst-plugin-spotify_0.9.0_armhf.deb
sudo ln -s /usr/lib/libgstspotify.so /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstspotify.so
```
A restart may be needed to get the plugin working. 

2. Now you should be able to test the gstreamer spotify implementation using the commandline interface:  
`gst-launch-1.0 spotifyaudiosrc username=<Username> password=<Password> track=spotify:track:3i3P1mGpV9eRlfKccjDjwi ! oggdemux ! vorbisdec ! audioconvert ! autoaudiosink`  

You need your regular credentials here not any API or application keys.

4. Download and extract the source code of the *mopidy-spotify* version created by Kingosticks. Here i used the branch *gstspotifysrc-hack*. **Please be aware this is work in progress and implementation may not be stable yet**  

```
wget https://github.com/kingosticks/mopidy-spotify/archive/refs/heads/gstspotifysrc-hack.zip  
unzip gstspotifysrc-hack.zip  
cd mopidy-spotify-gstspotifysrc-hack
```

5. Modify the source code you just downloaded to hardcode the credentials  
`vi ./mopidy_spotify/backend.py``

Almost at the end of the file you will find the following code:  
```
class SpotifyPlaybackProvider(backend.PlaybackProvider):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        username = self.backend._config["spotify"]["username"]
        password = self.backend._config["spotify"]["password"]
        self._auth_string = f"username={username}&password={password}"
````

I simply replaced the username and password logic with hardcoded values:

```
class SpotifyPlaybackProvider(backend.PlaybackProvider):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        username = MySpotify@email.com
        password = MyHardToBreakPassword123!
        self._auth_string = f"username={username}&password={password}"
```

6. Remove the exising installation of *mopidy-spotify* which is no longer working  
`sudo apt-get remove mopidy-spotify'

7. Compile and install the new *mopidy-spotify*  
```
cd..  
sudo python3 -m pip install .  
```

You may need to install pip to execute the compile command.  
This can be done via apt:  
`sudo apt-get install python3-pip`

That's it. After applying all this changes I was able to use the spotify feature of the phoniebox again.  
Even though there are some limitations right now which may especially affect those of you using phoniebox for Audiobooks.  
Currently the implementation only supports **Play** and **Pause** you cannot fast forward within a track and you cannot resume a track where you stopped. This hopefully comes at a later stage.  

My little boy is very happy now as the PhonieBox is back in action...

I appreciate any feedback if this was helpful and if the approach worked for you as well.

Hamsert, 2022-08-25:
> Hey all,
> 
> while this is not yet a perfect solution I got spotify working again on phoniebox. The solution is based on the thread referenced by @s-martin. Major thanks goes to all the contributors in the [https://github.com/mopidy/mopidy-spotify/issues/110 thread](https://github.com/mopidy/mopidy-spotify/issues/110).
> 
> **be advised: The solution requires some minor code changes and compilation of one software module** , so you should have some basic Linux knowledge to apply the workaround. I've applied the solution on a Raspberry Pi 4 and slight adjustments might be necessary on older PIs due to the different ARM-architecture.
> 
> What did I do to get the solution working:
> 
> 1. Download and install a precompiled version of the so-called spotify url handler for gstreamer. This is a specific version of the handler as it was adjusted to be mopidy compliant.
> 
> ```
> wget https://github.com/mopidy/mopidy-spotify/files/9176522/gst-plugin-spotify_0.9.0_armhf.deb.zip
> unzip gst-plugin-spotify_0.9.0_armhf.deb.zip  
> sudo dpkg -i gst-plugin-spotify_0.9.0_armhf.deb
> sudo ln -s /usr/lib/libgstspotify.so /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstspotify.so
> ```
> 
> A restart may be needed to get the plugin working.
> 
> 2. Now you should be able to test the gstreamer spotify implementation using the commandline interface:
>    `gst-launch-1.0 spotifyaudiosrc username=<Username> password=<Password> track=spotify:track:3i3P1mGpV9eRlfKccjDjwi ! oggdemux ! vorbisdec ! audioconvert ! autoaudiosink`
> 
> You need your regular credentials here not any API or application keys.
> 
> 4. Download and extract the source code of the _mopidy-spotify_ version created by Kingosticks. Here i used the branch _gstspotifysrc-hack_. **Please be aware this is work in progress and implementation may not be stable yet**
> 
> ```
> wget https://github.com/kingosticks/mopidy-spotify/archive/refs/heads/gstspotifysrc-hack.zip  
> unzip gstspotifysrc-hack.zip  
> cd mopidy-spotify-gstspotifysrc-hack
> ```
> 
> 5. Modify the source code you just downloaded to hardcode the credentials
>    `vi ./mopidy_spotify/backend.py`
> 
> Almost at the end of the file you will find the following code:
> 
> ```
> class SpotifyPlaybackProvider(backend.PlaybackProvider):
>     def __init__(self, *args, **kwargs):
>         super().__init__(*args, **kwargs)
>         username = self.backend._config["spotify"]["username"]
>         password = self.backend._config["spotify"]["password"]
>         self._auth_string = f"username={username}&password={password}"
> ```
> 
> I simply replaced the username and password logic with hardcoded values:
> 
> ```
> class SpotifyPlaybackProvider(backend.PlaybackProvider):
>     def __init__(self, *args, **kwargs):
>         super().__init__(*args, **kwargs)
>         username = MySpotify@email.com
>         password = MyHardToBreakPassword123!
>         self._auth_string = f"username={username}&password={password}"
> ```
> 
> 6. Remove the exising installation of _mopidy-spotify_ which is no longer working
>    `sudo apt-get remove mopidy-spotify`
> 7. Compile and install the new _mopidy-spotify_
> 
> ```
> sudo python3 -m pip install .  
> ```
> 
> You may need to install pip to execute the compile command. This can be done via apt: `sudo apt-get install python3-pip`
> 
> That's it. After applying all this changes I was able to use the spotify feature of the phoniebox again. Even though there are some limitations right now which may especially affect those of you using phoniebox for Audiobooks. Currently the implementation only supports **Play** and **Pause** you cannot fast forward within a track and you cannot resume a track where you stopped. This hopefully comes at a later stage.
> 
> My little boy is very happy now as the PhonieBox is back in action...
> 
> I appreciate any feedback if this was helpful and if the approach worked for you as well.

Thanks for this! Has anyone tried if this works on a pi 3B?

MrStrategy, 2022-08-25:
> Thanks for this! Has anyone tried if this works on a pi 3B?

Yep. Did this earlier today. Works!

84julian, 2022-08-28:
Many thanks for all to all the contributors! :-)

3 additional Points: 

1) I had some porblems cause i used the Spotify User Name and not the E-Mail.

2) the Email and Password must be in " "

3) the Spotify password is not allowed to have # in it cause of fstring.

It runs and i hope somebody will find a solution for playing playlists... At the moment there is a long time gab between the songs.

sebrep, 2022-08-28:
Has anyone tested this for the raspberry zero? My phonieboxes both sadly need the small version.

thomas-wolf, 2022-08-28:
Also working for me on a raspi 3b on buster. Trying this on some older phoniebox which still runs on stretch sent me directly into the dependency hell. Guess I will need to perform a dist upgrade first. 

Thanks to all contributers!

s-martin, 2022-08-29:
Glad to see the community came up with a workaround.
We will look into it and check, if that can be integrated and how stable it is.

> 5. Modify the source code you just downloaded to hardcode the credentials
> `vi ./mopidy_spotify/backend.py`
>
> Almost at the end of the file you will find the following code:
> ```
> class [...]

I'm not sure why this is necessary, though.

From my point of view the modified plugin should be able to grab the Spotify credentials from the usual config location. Maybe worth to look into that.

thomas-wolf, 2022-08-29:
> From my point of view the modified plugin should be able to grab the Spotify credentials from the usual config location. Maybe worth to look into that.

Guess I can confirm this, s-martin! I followed the steps of the workaround as described, but still phoniebox was not able to playback from spotify until I changed the credentials in the mopidy conf.

MrStrategy, 2022-08-29:
> It runs and i hope somebody will find a solution for playing playlists... At the moment there is a long time gab between the songs.

I read about delays before playback starts in the mopidy-spotify issue thread as well. At least for my RPI4 / RPI3B installation I can confirm playback, skip forward and skip back are working without any delay.

> From my point of view the modified plugin should be able to grab the Spotify credentials from the usual config location. Maybe worth to look into that.

Would be super if we could get it working like this. I gave it a lot of trial and error attempts to but always ended up with an mopidy error:

`error=GLib.Error('Could not get/set settings from/on resource.', 'gst-resource-error-quark', 13) debug='audio/spotify/src/spotifyaudiosrc/imp.rs(280): gstspotify::spotifyaudiosrc::imp (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstSpotifyAudioSrc:source:\nLogin failed with reason: Bad credentials'`

I also spent hours and hours in compiling a new version of gstreamer (1.20) as this is not shipped with buster and build process has a mass of dependencies. Finally I gave the *credential overwrite* a try and it worked immediatly.  
Some of you with more insights on architecture should get into this to find a better approach.

kingosticks, 2022-08-29:
A hash in the config file denotes a comment, i.e the text following the hash character is ignored. If I remember correctly, you can avoid that by surrounding your value in quotes:
```
password = "my#password"

There's no reason this would be different between the old (broken) version of Mopidy-Spotify and this hacked-up workaround version.

usamer77, 2022-08-30:
Works like a charme! Many thanks to @MrStrategy !!!

I just followed the steps, used my "USERNAME" not Email and "Password". I just was able to use the existing Phoniebox installation with all the associated cards, with no issues and no degradation of functions!

The only hint.. I'd like to recommend to be prepared while using part 2:

_```
Now you should be able to test the gstreamer spotify implementation using the commandline interface:
gst-launch-1.0 spotifyaudiosrc username=<Username> password=<Password> track=spotify:track:3i3P1mGpV9eRlfKccjDjwi ! oggdemux ! vorbisdec ! audioconvert ! autoaudiosink
```_

The test went ok, but with FULL volume and basically caused injuries while neighbors were falling out of their bed! ;-)

Cheers!

usamer77, 2022-08-31:
Modifying the 2nd Phoniebox also turned out to have a problem playing from Spotify until I changed mopidy.conf with the correct authentication credentials!

> Guess I can confirm this, s-martin! I followed the steps of the workaround as described, but still phoniebox was not able to playback from spotify until I changed the credentials in the mopidy conf.

If  you need to re-authenticate, here is the mopidy authentication link: https://mopidy.com/ext/spotify/#authentication

Just put in your client_id and client_secret, plus username and password and it should work!

CallMeErc, 2022-09-03:
> Hey all,
>
> while this is not yet a perfect solution I got spotify working again on phoniebox. The solution is based on the thread referenced by @s-martin. Major thanks goes to all the contributors in [...]

Wow, thank you very much for that workaround and sharing it here with lots of deperate parents, who had sad kids from the day on their boxes did not work anymore like they used to do.

When I followed your walkthrough, I noticed, that the following section was not in the code anymore:
> class SpotifyPlaybackProvider(backend.PlaybackProvider):
>     def __init__(self, *args, **kwargs):
>         super().__init__(*args, **kwargs)
>         username = [...]

I still tried to set it up following the other steps and also updated the mopidy.conf at "/etc/mopidy/mopidy.conf" via new authentication details. It did not work out.

I noticed, that the developer kingosticks has just worked on his project, you are pulling the source code from, and exactly that crucial piece of code changed. With that change, you cannot hardcode the username and password as described in your guide by pulling the newest branch commit.

So in order to follow the procedure of you, which results in a working setup, you need to change step 4 from :
> 4. Download and extract the source code of the _mopidy-spotify_ version created by Kingosticks. Here i used the branch _gstspotifysrc-hack_. **Please be aware this is work in progress and [...]

to 

> 4. Download and extract the source code of the _mopidy-spotify_ version created by Kingosticks. Here i used the branch _gstspotifysrc-hack_. **Please be aware this is work in progress and implementation may not be stable yet**
> 
> ```
> wget https://github.com/kingosticks/mopidy-spotify/archive/ed98d03e47d3b9c008d7463612124a90af5cf017.zip  
> unzip ed98d03e47d3b9c008d7463612124a90af5cf017.zip  
> cd mopidy-spotify-unzip ed98d03e47d3b9c008d7463612124a90af5cf017 

This will pull the commit that was the leading one, when the workaround guide was published.
With that special commit everything worked like a charm.
So thank you very much!!!

Kind regars
Eric

tulpenwase, 2022-09-21:
Hello people,

I have a question about the hack written here by @CallMeErc  and @Hamsert. 
I can't get the hack to work for me. I can not play Spotify songs or album in Mopidy as well as in Phoniebox.
I can only call it from the command line with gst
`gst-launch-1.0 spotifyaudiosrc username=<Username> password=<Password> track=spotify:track:3i3P1mGpV9eRlfKccjDjwi ! oggdemux ! vorbisdec ! audioconvert ! autoaudiosink`
Which Mopidy and Phoniebox versions do you use that it works?
And can you play whole albums with gst?
My version is:
Pi 3B
Phoniebox - 2.2 - 305325d - master

thanks a lot

Baggerfahrer, 2022-10-09:
> Hello people,
>
> I have a question about the hack written here by @CallMeErc and @Hamsert. I can't get the hack to work for me. I can not play Spotify songs or album in Mopidy as well as in Phoniebox. [...]

for me the above mentioned guide does also not work. i think i have a similar issue like @tulpenwase 
i followed the tip from @CallMeErc as well.

Mopidy seems not to run, even after the installation of the new mopidy hack. In Iris, i see the following info-box.
[image]

Maybe an idea for instruction could be:
`pi@raspberrypi:~/mopidy-spotify-ed98d03e47d3b9c008d7463612124a90af5cf017 $ sudo systemctl restart mopidy.service
pi@raspberrypi:~/mopidy-spotify-ed98d03e47d3b9c008d7463612124a90af5cf017 $ sudo journalctl -f -u mopidy
-- Logs begin at Sun 2022-10-09 05:52:58 BST. --
Oct 09 06:04:01 raspberrypi systemd[1]: Stopping Mopidy music server...
Oct 09 06:04:01 raspberrypi mopidy[10478]: INFO     [IrisFrontend-10] mopidy_iris.core Stopping Iris
Oct 09 06:04:01 raspberrypi mopidy[10478]: INFO     [MainThread] mopidy.commands Stopping Mopidy core
Oct 09 06:04:01 raspberrypi mopidy[10478]: INFO     [MainThread] mopidy.commands Stopping Mopidy backends
Oct 09 06:04:01 raspberrypi mopidy[10478]: INFO     [MainThread] mopidy.commands Stopping Mopidy audio
Oct 09 06:04:01 raspberrypi mopidy[10478]: INFO     [MainThread] mopidy.commands Stopping Mopidy mixer
Oct 09 06:04:01 raspberrypi systemd[1]: mopidy.service: Succeeded.
Oct 09 06:04:01 raspberrypi systemd[1]: Stopped Mopidy music server.
Oct 09 06:04:01 raspberrypi systemd[1]: Starting Mopidy music server...
Oct 09 06:04:01 raspberrypi systemd[1]: Started Mopidy music server.
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.1.1
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Oct 09 06:04:05 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.config Loading config from command line options
Oct 09 06:04:05 raspberrypi mopidy[14746]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, mpd, http, iris, spotify, local, stream, m3u, softwaremixer
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Oct 09 06:04:07 raspberrypi mopidy[14746]: ERROR    [MainThread] mopidy.commands Got un-handled exception from SpotifyBackend
Oct 09 06:04:07 raspberrypi mopidy[14746]: Traceback (most recent call last):
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/commands.py", line 241, in _actor_error_handling
Oct 09 06:04:07 raspberrypi mopidy[14746]:     yield
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/commands.py", line 406, in start_backends
Oct 09 06:04:07 raspberrypi mopidy[14746]:     config=config, audio=audio
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/pykka/_actor.py", line 91, in start
Oct 09 06:04:07 raspberrypi mopidy[14746]:     obj = cls(*args, **kwargs)
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_spotify/backend.py", line 24, in __init__
Oct 09 06:04:07 raspberrypi mopidy[14746]:     self.playback = SpotifyPlaybackProvider(audio=audio, backend=self)
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_spotify/backend.py", line 46, in __init__
Oct 09 06:04:07 raspberrypi mopidy[14746]:     username = self.backend._config["spotify"]["vitaminrv@gmx.de"]
Oct 09 06:04:07 raspberrypi mopidy[14746]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/config/__init__.py", line 312, in __getitem__
Oct 09 06:04:07 raspberrypi mopidy[14746]:     item = self._data.__getitem__(key)
Oct 09 06:04:07 raspberrypi mopidy[14746]: KeyError: 'vitaminrv@gmx.de'
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Oct 09 06:04:07 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Oct 09 06:04:08 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Oct 09 06:04:08 raspberrypi mopidy[14746]: INFO     [HttpFrontend-9] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Oct 09 06:04:08 raspberrypi mopidy[14746]: INFO     [IrisFrontend-10] mopidy_iris.core Starting Iris 3.45.0
Oct 09 06:04:08 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Oct 09 06:04:08 raspberrypi mopidy[14746]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Oct 09 06:04:09 raspberrypi mopidy[14746]: INFO     [MpdSession-12] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56862

`

Many thanks!

[76 more replies]

closed 2024-04-14 by AlvinSchiller