MiczFlor/RPi-Jukebox-RFID #1824: Nothing works after reinstall of 2.

baschior, 2022-05-28:
I installed the jukebox several times and everything was fine.

This time no rfid reader functionality, no volume control, no spotify while there was no hardware change at all....

What happend to the Project?

s-martin, 2022-05-28:
Several issues have been merged, but not to master yet. You could fix these issues manually for the moment:
- https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791
- https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1789

Unfortunately Spotify disabled also some things of their API, which breaks the library mopidy-spotify we are using. This will only be fixed in the upcoming 3.x release. See here for details: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1815

baschior, 2022-05-28:
Thanks for the quick answer, Im not fit enough to fix it manually, so I leave the box aside and wait for 3.0 sadly

s-martin, 2022-12-18:
V2.4.0 was released today - fixed.

closed 2022-12-18 by s-martin